prusa3d/Prusa-Firmware #1796: MK3S fails to unload filament during MMU2S tool changes, tools are off for remainder of print

cjbaar, 2019-05-05:
I am not even sure if this is one issue or two. I have had so many types of tool changes failures with the MMU2 I cannot count them all. The latest firmware updates and IR sensor have corrected many of the physical problems with MMU operations, but I am now experiencing a new type of failure on a frequent basis.

I have a hard time catching this in the act, so I am not sure the order that things fail, but what often seems to happen is the MMU is reset. I do not know if this is the source of the issue, or a side effect of it. But the real issue is that the extruder does not unload the filament. The IR sensor is triggered, so the printer should know full well that filament is still in the extruder, but it does not make any attempt to release or eject it. Therefore, the MMU just spins and grinds out the filament, and eventually triggers an unload failure.

So, problem #1: the extruder is not releasing filament

Here is where it gets even weirder. After the failure, sometimes that particular tool gets ignored for the rest of the print. In order words, every time it is supposed to use F1, it instead uses F2, and the print is now ruined. F2 still uses F2, but so does F1. When it should be laying down filament 1, the printer states "F2". When it is then supposed to transition from F1 to F2, it starts the ramming, retracts a bit, then purges like it switched even though no filament change was performed.

I do not know if this should be a separate issue. I am attaching a picture of the recent failed print. All of the "stringy" green (still unsure why flow is affected) should be black.
[image]

cjbaar, 2019-05-14:
A bit more info. I had this issue reoccur, this time after a resumed print the printer began substituting F3 for F2 for the remainder of the print. However, I then tried a new print (different gode), and the substitution continued through that print as well. So somehow the glitch gets held in the printer memory.

mkbel, 2019-05-20:
Do you have feature spool join turned on? It looks like spool join is turned on and filament run out is detected and next spool is used permanently.

cjbaar, 2019-05-21:
Thanks @mkbel. That sounds plausible. I will check out the settings.

Mirarkitty, 2019-06-11:
The spool join feature is causing all kinds of weird behaviour, but probably because something else determined the filament had run out... Now, why it does that...

closed 2023-09-28 by github-actions[bot]