prusa3d/Prusa-Firmware #1797: First Layer Height..

racoutlaw, 2019-05-05:
The first layer height on my i3 mk3s will not get below a .15 first layer.. if i set first layer to .1 or even a .05 for first layer the first layer will always be .15 no matter what i have it set to print it at. Customer support chat informed me to make a post here about this issue as it does the same thing on his printer when using the file i sent over...

leptun, 2019-05-05:
Hi,

What are your slicer settings?

You have to change it from "Print settings" tab on the left like in the first photo in order for it to actually change.
[image]

If you change it like in the second photo the only thing that changes is the layer height on all other layers other than the first
[image]

Nippylicious, 2019-05-05:
Somewhere in the manual...

1st layer ist always 0.20
No matter witch height da you print at (0.1, 0.2, 0.3...)

racoutlaw, 2019-05-05:
If you change first layer height in your response @leptun from the .2mm that it shows for first layer height to say .1 or .05 the lowest the printer will go is down too .15 for the first layer height and then what you have set at in the Layer Height takes over after that and prints...   So if i set say Layer Height too .05 and First Layer Height too .1  when i start printing the first layer will be .15 and the rest will be .05. Printer will not do a first layer height below a .15....

racoutlaw, 2019-05-06:
Firmware locked cant go lower than .15 for first  layer

closed 2019-05-06 by racoutlaw