prusa3d/Prusa-Firmware #1812: Ignored gcode lines. Circle segment reduction?

Swesen, 2019-05-10:
I have a strange problem where my MK3S with MMU2S will just skip a couple gcode commands. I suspect this could be something where the firmware is trying to reduce the number of segments in a circle.

A part that produces this (at least this latest iteration) is a triangle shaped part with one right angle and the right angle is rounded. I tried via SD-card and octoprint, and slic3r PE 1.41.3, 1.42 beta1 and 2 to rule out any other component. This also happened with another part that had a large circular perimeter. Also the same part with the hole in the middle slightly more to the center had no problem printing...

I also tested to reduce the face count on the exported STL from fusion 360. Which helped on one of the parts but not the other.

[image]

[image]
[image]
[image]
[image]

leptun, 2019-05-10:
Please post the gcode as I want to test this on my mk3

Swesen, 2019-05-10:
The one that say hidef is the high face count one that both fails on and the other one is where one fine one fail gcode. I took a video on this problem also, will upload shortly.
[Gangjarn v2_0.2mm_PET_MK3SMMU2S (2).zip](https://github.com/prusa3d/Prusa-Firmware/files/3167389/Gangjarn.v2_0.2mm_PET_MK3SMMU2S.2.zip)

Swesen, 2019-05-10:
Recorded from after purge line but I zoom in at the problem at around 1 min https://youtu.be/VxURmfWbbfs

Swesen, 2019-05-11:
This seems like some random firmware/EEPROM corruption that happened on my einsy board. After flashing the same firmware(the same file that I uploaded 1 month ago when I upgraded my MK3 to MK3S with MMU2S) that had the problem it now prints these files without problem...

closed 2019-05-11 by Swesen