prusa3d/Prusa-Firmware #1815: Selector should be prevented from moving while FINDA triggered

PxT, 2019-05-10:
MK3S + MMU2S

It seems the printer can sometimes get into a state of confusion when a filament load or unload goes wrong and it will end up moving the selector while the previous filament is still partially or fully engaged.  Of course this just snaps the filament and then there is no chance it will recover.  I checked the firmware and I couldn't find any failsafe to prevent the selector moving while FINDA is engaged?  I'm new to the MMU variant of the firmware so maybe I missed it.

If not already there then I think there should be a check -- if FINDA == triggered, then prevent selector change commands and go to "needs attention" mode.

WarlockD, 2019-05-13:
I second this.  My button on the MMU was shorted so when it changed it moved one more notch over.  Honestly I think a better protocol needs to be implemented (like RFC 916 is what I am experimenting with) that clearly tells both MMU and the printer the channel is open.

Honestly, I just think we need to make the MMU more "dumb".  I kept wanting to extend the light/button interface to something external to my enclosure till I thought, "Hold on, doesn't the printer have complete control though the serial port?"  I mean hell, its using scanf, of all thing, to read 2 byte commands.  And the number of shift commands throughout all the multiple loops.  I think we need to change the idea of this is just an "addon" and more "extension".  Atleast that's my thinking.

mkbel, 2019-05-20:
MMU keeps track of filament presence in variable isFilamentLoaded defined in
https://github.com/prusa3d/MM-control-01/blob/master/MM-control-01/mmctl.cpp
It is initialized during power up in setup() function to FINDA reading.
For rest of the time, it is updated when unload_filament_withSensor() and load_filament_withSensor() finishes. Those function in theory doesn't terminate unless right FINDA reading. If you find some logical error or reproducible situation, feel free to report it.

PxT, 2019-05-20:
One thing I notice is that there is more than one variable used to determine the "loaded" state.  For example in try_feed_filament() it uses its own private _loaded variable.  I think this function can alter the load state without updating isFilamentLoaded ?

In my opinion the select_extruder() function in mmctl.cpp should also check isFilamentLoaded directly.  The function description says that "caller is responsible for ensuring that filament is not loaded" but since this is a potentially destructive operation it would be be good practice for the function to do a double check itself as a fail-safe.  Trust, but verify.
 

See https://www.facebook.com/photo.php?fbid=10213143046201092&set=p.10213143046201092&type=3&theater for an example showing the destructive action. (not my photo)

Panayiotis-git, 2019-05-20:
The simplest test:
remove the orange PTFE from the MMU
insert a filament from the front side so it triggers the FINDA
select Load All from the menu
Unexpected result: The selector will move from the 1st to the 5th position like nothing strange is happening.
Expected Reasonable result: After the 1st position the selector should freeze right there and a strong indication that something is wrong like blinking all the red lights because the FINDA is still triggered.

This check should be done before each movement of the selector. It should not kept in any variable.

sticilface, 2019-07-06:
I second this..  

Selecting from the menu "load to nozzle" if a filament is already loaded causes the selector to move and bend / break things.  

I've ripped out the knife blade doing this.  

there should be a check in the selector mover to check finda, and in the load to nozzle there should be an unload move first if finda is triggered.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

Selector should now never move when the FINDA detects a filament.

closed 2023-07-31 by gudnimg