prusa3d/Prusa-Firmware #1816: MM3S/MMU2S Under-extrusion upon leaving wipe tower

cjbaar, 2019-05-11:
I have had this issue since I first received the MMU2 last year. No recent software/firmware updates seem to have addressed it. Whenever the printer goes through the purge process after changing filaments, it either over-retracts or under-primes leaving the wipe tower, and the first extrusion is significantly low (several mm). This happens on every tool change, regardless of filament or layer.

Is there a setting for this in Slic3r PE? If so, I am not seeing it. There seem to relatively few settings for wipe tower speed, flow, etc.

cjbaar, 2019-05-11:
This isn't the greatest picture, but you can see where the first few items are under-extruded.
[image]

cjbaar, 2019-05-11:
Another (mediocre) pic after the print finished. You can see the poor extrusion on the first white and dark grey pixels. There are around 6,500 pixels in this print (1.8mm^2), and the first off the wipe tower are the only ones that look like this.
[image]

cjbaar, 2019-05-14:
Another example I see on pretty much every print. This is the first layer, wipe tower on the right, and you can see the gap in the skirt (about 8mm).
[image]

cjbaar, 2019-05-14:
2nd layer, first tool change, same under-extrusion
[image]

bynnek, 2019-05-18:
I've seen this issue as well and just assumed I had something set wrong.  Interesting to see someone else with the same problem.

cjbaar, 2019-08-29:
I re-opened this in the slicer forums:
https://github.com/prusa3d/PrusaSlicer/issues/2834

closed 2019-08-29 by cjbaar