prusa3d/Prusa-Firmware #1817: Preheat error

Gountled, 2019-05-12:
Using I3 MK3S and MMU2S:
Printer FW 3.7.0-2201
MMU FW 1.0.5-297

My 5hr. print aborted at 89% wit preheat error (not bed preheat error!)
This time, all my PLA filaments used the sam temperature and this happened during a change between Filament 4 and 5. The printer fully loaded the filament no. 5 but than stopped wit preheat error. No chance to restart printing! The print is lost!

This happened second time to me, but last time there was a temerature difference in the filament used.

In addition, I would like to commend the hot temperature of the extruder stepper motor. It went up to 62 degrees centrigate.

[image]

leptun, 2019-05-12:
Maybe this issue is related since you have the R4 extruder #1746

Gountled, 2019-05-12:
I never had this problem before I got my MMU2S and the printer has been upgraded to the latest Extruder design together with the MMU.
If this is really related to the new extruder design, this would be a shame.
Reducing the fan speed is definitely not a solution as it has a direct impact on print quality.

Nevertheless I will give the Silicone sock a chance.

Gountled, 2019-05-14:
Update:
I attached the silicone sock to my Heatblock and was able to perform an 11hr print with 263 filament changes. Therefore thanks for the hint!

Still the extruder motor is getting very hot up to 60°C. This already cuased some problems with PLA getting wrapped around the bond tech gear. For the time being I attached a heatsink on top of the extruder motor.

closed 2019-05-14 by Gountled