prusa3d/Prusa-Firmware #1829: Still getting mmu2s load failed 3.7.1

dude2k5, 2019-05-14:
Is there any way to get more diagnostics on why it is failing when you get this message? 

I got this error again today, it seemed to be going alright, 5% in the print and then it tried to do something and failed. 
But I can't figure out why or what failed. The IR sensor looks ok. The filament had the previous color when i pulled it out (melded together), so im guessing it tried to push but got stuck, but I cant see where. It did a few layers no issue. Is there something we can enable that shows what exactly failed?

Also previously you could at least fix the problems, the printed paused and waited for you to fix it, but it just says failed so you have no chance of recovery. And if you dont know where the problem is, you cant fix it.

Gountled, 2019-05-14:
The blinkink frequency shows in which phase the error occured, as written in chapter 13.2 of the MMU handbook:

● Slow flashing
​ - fault during filament load

● Rapid flashing ​
- fault during filament unload

michalxfanta, 2019-05-15:
Are the LEDs flashing slow or rapid?

dude2k5, 2019-05-16:
headpalm.jpg

So Ive tried to print the model 3 times. First time it failed around 5%, got the "mmu load failed", but I was unsure if it was flashing faster/slower (i had not created this ticket yet). However the filament was in the tube going to the extruder. When I pulled it out, the old color was attached to the new one, so it acted like it was pushing it toward the extruder, but it never got far enough to push the old color out. 

So I tried again, same print, didnt change much except cut the filament sharp again. It failed once again, almost in the same exact place. Im only doing 2 colors in this print, and it's doing 5-6 layers fine, between both colors, but then something gets stuck. "MMU load failed" once again. I then created this ticket. Again, I didnt know the flashing led diagnostic, so I attempted to print again to see if it was flashing faster/slower this time

So I tried again last night, didnt do anything to it, didnt even cut it sharper, i just unloaded the filament (it was stuck inside from before, when I pulled it out it had the old color like last time) and then tried to reprint. This time it's 40% done and still going. Im not even sure how, since the other 2 got stuck pretty early on. 

Can we perhaps keep this open until the end of this print? I can then close it if it succeeds, this thing gives me a headache sometimes lol.

dude2k5, 2019-05-16:
Came back from work, did it again. This time at 43% done. Did many layers fine, then got stuck randomly. Can't even really fix it, won't let me. 

The flashing was fast, so it was trying to unload it seems

If I pull it out, the filament is worn down at the mmu portion, it tried to eject it but couldnt. Got too worn down by trying too long. No idea why it was fine for so long then decided nope.

Sadly if I tried to continue on the mmu by pushing the button (orange flash went to orange/green when i cleared the jam), it didnt let me continue the print, it just quit

https://imgur.com/a/LDViRId

mkbel, 2019-05-20:
Clicking printer knob should resume printing. Printer will retry last failed filament load action. If it fails again it allows you again retry the upon printer knob click.

macgaver, 2019-05-23:
I have the exact same random issue. Using MMU2S with MK3S, all original, no mods. All latest firmware.
Thanks for the tip that we can click the nob to continue, that part work. It helped me try over and over to continue a print (compared to restart all over again like I did the 5 times before).

When it try, I can ear the clicking noise from the MMU2, it try to push the filament faster than the extruder pull it. At the beginning of a print it doesn't do that, but it look like after filament change, or error/retry, it get worst.  I can help it pass removing tension on the mmu2 springs just at the moment where it reach the extruder ... but 2 minutes after when it's time to change filament again, it is to be done again.

I decided to print a small mmu2 stress file (5 colors changing every minutes). It's in progress now if you guys want me to test something. I will print this thing until I debug everything and can trust the mmu2.

closed 2023-09-28 by github-actions[bot]