prusa3d/Prusa-Firmware #1830: MK2.5S Firmware 3.7.1 - Z calibration issues

xyipher, 2019-05-14:
Unable to get Z calibration or XYZ calibration to work since firmware upgrade. 

Checked pinda probe, probe is working. 
Adjusted pinda probe multiple times. 
PEI sheet is not on the bed.
Checked the Z-Axiz screw covers on the top of stepping motors. 

Move all the way up to the top of the Z axis until it tops out. Place paper on bed, and hit continue. Head moves all the way down until I see pinda probe light go out then error shows on screen. Pinda probe is completely outside of circle. Picture attached. 

Error
Calibration failed! Check the axes and run again.

No matter what I do I can not get it to pass point 1 on Z axis calibration. Did not have a problem before the firmware upgrade. 

Running the upgraded extruder R3 (B7).
Firmware 3.7.1-2266

[image]
[image]
[image]

leptun, 2019-05-14:
You have the MK2.5S printer, but you said you have R3 extruder. This is not possible. You must use the r4 extruder with Mk2.5s firmware or printer.

You should upgrade the extruder to r4 or downgrade the firmware to 3.7.1 for mk2.5 not S variant

leptun, 2019-05-14:
R1, R2, R3 (upgraded extruder) are for Mk2.5/Mk3

R4 is for Mk2.5S/Mk3S only.

Same applies for firmware

xyipher, 2019-05-14:
It must be the R4. I was trying to remember the one part from memory. Its the upgraded extruder with the fan at the 45 degrees. It was working fine until this most recent firmware upgrade.

leptun, 2019-05-14:
Could you please make a video where you show the support menu and how the xyz calibration fails?

xyipher, 2019-05-15:
Went through my records. I upgraded from the MK2S -> MK2.5 -> R3 extruder design. I was looking into upgrading to the MK2.5S but haven't yet. Must have been why I was thinking MK2.5S. 

Downgraded the firmware and calibration works fine.

Thanks for the clarification.

closed 2019-05-15 by xyipher