prusa3d/Prusa-Firmware #1831: 3.7.1 Language Issue?

Crazy-Crosshair, 2019-05-14:
I have a problem on My MK3 (non S) with 3.7.1 i am in a Loop when the printer asks me to choose a Language. If i choose German he will ask me again and again. If i choose English everything is fine.

leptun, 2019-05-14:
Try factory resetting the printer deleting all settings

Link: https://help.prusa3d.com/article/SYvbQ66IXF-factory-reset

Crazy-Crosshair, 2019-05-14:
I resetted the Language. Same Problem. 

Point to Deutsch  -> Copy selected language? -> Yes - Original Prusa i3 Prusa Research Language update... -> List of Languages to choose frome

leptun, 2019-05-14:
Try reflashing 3.7.1 and if it still fails try 3.7.0. Also is this compiled by you or downloaded from the releases?

Crazy-Crosshair, 2019-05-14:
It is right from your Homepage (https://www.prusa3d.de/treiber/)

Crazy-Crosshair, 2019-05-14:
Reflashing to 3.7.1 worked for me!

leptun, 2019-05-14:
Great to hear.

closed 2019-05-14 by Crazy-Crosshair