prusa3d/Prusa-Firmware #1833: MMU2S not loading filament to the nozzle

zeroflow, 2019-05-14:
**Whats the problem:**

Upon trying to load the filament to the nozzle, the filament is loaded not all the way to the nozzle.

It triggers the FINDA, but does not continue any further.

**Steps to reproduce:**

Option 1: Send T1 from the Terminal
Option 2: Select "Load to Nozzle" on the printer

**Expected result:**
1) MMU2S switches filament slot
2) The selected filament starts a slow move beyond FINDA
3) Starting from PINDA, the filament is loaded fast near to the bondtech gears
4) It is then loaded correctly into the nozzle

**Observed result:**
1) MMU2S switches filament slot
2) The selected filament starts a slow move beyond FINDA
3) The filament feeding stops at the height where the coupler for the PTFE tube to the hotend assembly sits
( 4) when printing normally, now the printer rports "loading failed")

When flashing

zeroflow, 2019-05-14:
OSI Layer 8 issue: Seems like the IR-Sensor alignment was off

closed 2019-05-14 by zeroflow

leptun, 2019-05-14:
Was it always triggered? This is what happened to me when I upgraded to Mk2.5S. The solution was putting a small piece of zip tie between the pcb and plastic part towards the front and screwing the pcb like before.