prusa3d/Prusa-Firmware #1836: MK3 Under Extrusion after travelmove

rennne, 2019-05-14:
After some travel moves it happens that the printer is unable to get extrusion going immediately.
So there is then a bit of plastic missing. It usually takes 5 to 10 mm then everything is fine again.

I have seen that happening on our Prusa in the company -> and older mk3 with a noctua fan and I think ICE PLA fialement.
and also
on my private mk3 ->( a bit newer with the black fan) with green Prusa PLA.

rennne, 2019-05-16:
I should add:
It happens only with a specific kind of print:
So far I have seen that only on  objects thin walls ( 2mm thick ), like pipes or boxes with a few holes.

mkbel, 2019-05-20:
It would help if you attach problematic gcode so we can reproduce.

rennne, 2019-05-20:
The GCode:

[SpurGear48_0.15mm_PLA_MK3.zip](https://github.com/prusa3d/Prusa-Firmware/files/3199253/SpurGear48_0.15mm_PLA_MK3.zip)

The result:
[image]

closed 2023-09-28 by github-actions[bot]