prusa3d/Prusa-Firmware #1837: 3.7.1 MMU2s filament grinding during first load

bynnek, 2019-05-15:
Printer: MK2.5s with MMU2s
Firmware: 3.7.1
Issue: Issue is happening when printing in single material mode, the Tx command is invoked and the filament is chosen on the screen at the start of a print.  What I expect to happen next on this first load is the extruder would start spinning and the filament starts traveling to the extruder from the MMU2s and stops once the filament sensor detects the filament has reached the extruder.  What is actually happening is the extruder motor does not spin.  Filament travels from the MMU2s down to the extruder, binds up because the bondtechs aren't spinning in the extruder and then stops after some grinding.  Once the nozzle and bed have fully heated and the mesh bed leveling has finished, it retracts the filament all the way to the MMU2s and then attempts to load again.  This second time, the extruder spins correctly and filament is loaded properly.  

Seems like for some odd reason the first load is not spinning the extruder motor and it isn't until the second attempt at loading filament does it actually spin that motor.  I have not noticed this when doing a multi material print.

RakNoel, 2019-05-18:
Having the same issue, with my setup aswell as some other issues. Might be related to both #1846 and #1847

mkbel, 2019-05-18:
Root cause of this issue is, that movement of bondtechs is disabled in some low level routine if nozzle temperature is not high enough. As priority of this issue low, it can take long time to fix. If it bothers you, as an workaround you can calibrate bowden lenght for each filament on MMU unit so it is not slamming against stopped bondtechs.

bynnek, 2019-05-18:
Thanks for the explanation, that makes a lot of sense.  Maybe I'll modify the gcode so the Tx command comes after the nozzle heating has finished.

nz2o, 2019-06-08:
I'm unfamiliar with github's system itself, so forgive me if I am using it incorrectly. I am having the exact same issue (and some unrelated issues). Not using Octoprint, however. 

mkbel's explanation makes sense, and did resolve part of the problem. I had to disable crash detection for a more reliable load, however. (I've been in touch with the support team.)

gudnimg, 2023-07-24:
Just adding this to the MMU milestone as a reminder to close this issue.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg