prusa3d/Prusa-Firmware #1842: MMU2S load failed

rickyh777, 2019-05-16:
Weird one here. MMU Load is failing (MMU2S and MK3S). When attempting to load to nozzle from LCD screen, it tries one and I can see it catch the bondtech gears. The bondtech gears then retract past themselves before the MMU retracts the filament. It tries again and the printer thinks it successfully loads asking me 'is the correct filament extruding' even though the MMU retracts the filament back to unloaded position. The MMU however blinks slowly red. When loading from a print however, exact same symptoms in terms of actual mechanical movements, but the LCD screen reports MMU load failed while the MMU shows a green light.

mkbel, 2019-05-18:
Do you hear clicking noise at the moment extruder is moving down, before it starts retracting? There is check to test if filament goes freely to heat-break. Bondtechs move down and up nearly the same distance. If the filament crashes into heat-break down distance is smaller then up distance leading to filament jumping out of bondtechs in the end of movement and this is detected as MMU load failed.

mkbel, 2019-05-18:
If your hot-end has different geometry, this could be failing all the time. In this moment we have no parameter to tune this. If you need to change this distance you need to tweak source code. https://github.com/prusa3d/Prusa-Firmware/issues/1689#issuecomment-481453003
You also probably need to tweak cooling moves in Slic3r - there are several parameters for this. And you need elastic tube in your hot-end with inner diameter smaller than nominal diameter of filament in order to shape thin enough tips.

iconoclast, 2019-05-19:
I'm seeing this issue with the stock extruder and hotend (MK3s upgraded from an MK3 via the MMU2s kit).  
I initially started experiencing the mmu loading problem described in https://github.com/prusa3d/Prusa-Firmware/issues/1689 
After wrestling with that for awhile I just decided to do a few single color prints instead using the load-to-nozzle option only to run into the behavior described in the first post of this issue.  The only difference from the issue description is that I just have the stock printer configuration.

rickyh777, 2019-05-19:
The mosquito hotend should have the same geometry with the exception of the distance between the gears and the nozzle being longer. I do hear a clicking however it seems to be due to the RPM of the direct drive being much faster than that of the bowden style from the MMU. It loads to nozzle fast but seems to slow down before actually reaching the nozzle. One thing I didn't consider was attempting an individual tube length calibration so it makes contact with the direct drive at about the same speed (I think). Will that have any affect on anything?

rickyh777, 2019-05-20:
That actually seemed to be the problem, at least in 3.6 which I reverted to. It may also fix the problem in 3.7, but basically since the extruder gears were moving at a different speed from the MMU gears, the extruder would immediately recognize a 'Jam' as it couldn't pull the filament which was entering at a much slower speed. (I have a custom PFTE tube length, not the one that came in the box due to a weird setup)

sticilface, 2019-06-19:
> Do you hear clicking noise at the moment extruder is moving down, before it starts retracting? There is check to test if filament goes freely to heat-break. Bondtechs move down and up nearly the [...]

This seems to be happening to me all the time.   I have a MK2.5 that was upgraded to MK2.5S with MMU2S.  I have quadruple checked everything.. if i disconnect the PTFE tube between the MMU and the printhead and just use some filament it works perfectly, there is no resistance it just feeds in beautifully...  if i then use the MMU I get a clicking sound and it fails every time for filament 5, some times for the other ones.  I've cut perfect tips..  I don't know what else I can do.  I've stock everything.  

Could this be the bond tech gears in the head trying to pull the filament too hard from the MMU and causing the grinding then it has not pulled it in far enough so that when it does the inout bit it pops out of the printhead bond techs, the IR sensor goes to 0 and it fails.. maybe a slower loading speed? 

This is 100% reproducible with filament 5 my other ones fail sometimes..  (which i don't understand.).  I've chopped it several different ways, checked alignment of the MMU + bearings...  

any ideas?

mkbel, 2019-06-19:
@sticilface: Did you get new heat-break with your upgrade kit "MK2.5 to MK2.5S with MMU2S". If not, this could be the problem, as MK2 heat-break has different leading edge than MK3 heat-break. If you didn't get heat-break with your upgrade kit, ask Prusa support for new heat-break. I expect that you should get it for free.

sticilface, 2019-06-19:
I did not replace the heat break... so I will ask

skylord73, 2019-08-04:
Same problem.
I'm using filament 5 (PLA) with a single filament setting in slic3r 1.4.13
the first print (with the filament tip cut at 45°) works fine, the second print with the same filament, reached the extruder, I see the IR triggering (idle moves and hear the noise) but the printer unloads the filament, tries again and the stopped with MMU loading issue.
I eject the filament, cut the tip again and everything works fine (10 test done)

rickyh777, 2019-08-05:
> Same problem after upgrading to MK3s and MMU2S (kit).
> I'm using filament 5 (PLA) with a single filament setting in slic3r 1.4.13
> the first print (with the filament tip cut at 45°) works fine, the second print with the same filament, reached the extruder, I see the IR triggering (idle moves and hear the noise) but the printer unloads the filament, tries again and the stopped with MMU loading issue.
> I eject the filament, cut the tip again and everything works fine (10 test done)

Sounds like you need to tune your cooling passes if it works the first time but not after its been retracted. The tips are probably too 'globby' so they get jammed as they enter your heat break which is highly toleranced. The extruder detects the crash and pulls it out and tries again. Check out the instructions in troubleshooting in the mmu2 guide (I don't have the link on hand) as it explains how to do this step quite well

closed 2023-09-28 by github-actions[bot]