prusa3d/Prusa-Firmware #1844: MK3 Layer Shifts with Octoprint for Firmware 3.7.1

6ill9re9, 2019-05-17:
Even when I pinch the X-axis it takes several attempts to detect a crash...
Not cool.

leptun, 2019-05-17:
Try doing selftest as during the test crash detection is calibrated.

6ill9re9, 2019-05-19:
Thx, I will give it a try!

leptun, 2019-05-19:
If this doesn’t solve your problem:

Check wether the belts are loose and skipping teeth of the belt, not steps of the motor. Also check if the pulleys are loose on the motor.

If it is not a hardware problem, then try reflashing the firmware and doing factory reset.

You said you upgraded to 3.7.1. What was the previous version and does this problem occur on the old version? Sometimes stuff gets corrupted when jumping from an old version to a new one.

Boym323, 2019-10-31:
I have the same problem on my MK3S. I checked belt, lubricated all three axis, flashed last firmware 3.8.1 and I did factory reset. But still when I print from Octoprint I get randomly shifted layer X or Y. But when I print the same G-code from SD cart all parts are OK.

closed 2023-09-28 by github-actions[bot]