prusa3d/Prusa-Firmware #1845: Feature Request MMU2S: MMU Needs Attention

vertigo235, 2019-05-17:
It would be super helpful if the PINDA and IR Sensor status would be shown on the MMU Needs Attention Screen.

On that note, it would be super helpful if this indicator could be worked into the main screen as well.  I know there is a screen burried in support, but you can't get to it when the MMU is in "Needs Attention" mode.

kmshort, 2019-05-18:
+1 Agreed. The user needs all the help they can get, to try and decipher what on earth has triggered the problem.

mkbel, 2019-05-20:
This is nice to have feature, but with current communication protocol with MMU and MMU software architecture it is impossible to implement. This will take some time.

kmshort, 2019-05-20:
I don't understand why the system wasn't that intelligent from the outset. However, I'm not a core developer so I do not know the limitations of the hardware. It definitely needs work though. The whole loading/unloading IR/Finda communication system needs an overhaul. There are so many people out there with issues, the prusaprinter forums trouble shooting section is full of loading/unloading sensor issues.

vertigo235, 2019-05-20:
I think the most useful piece of data would be the IR sensor status, currently we get the temperature and set temperature status on this screen.  The IR sensor is MK3 side, is this status not possible to show here?  I can understand why the MMU PINDA status may not be as practical.

frankracis, 2019-08-29:
If your software architecture doesn't allow for proper error handling, then fixing it needs to be a priority.

It's ridiculous that you need to resolve errors via blinking lights, special sequences of buttons, and random guessing on a device that has an LCD screen and rotary knob.

It would be one thing if errors were rare, but even in Prusa's own testing, over 25% of print jobs encountered an error (7% failed completely, 19% required intervention).   This level of reliability makes error diagnosis and recovery critical.

There are several other error handling tickets out there, but it seems that the inability for the main board and MMU to communicate during an error state is a common blocking issue.

vintagepc, 2020-11-02:
Linking to duplicate #1845

gudnimg, 2023-01-30:
I realise this is a 4 year old issue at this point but am adding this to the 3.13 milestone as a reminder for us to close the issue. The well known “ MMU Needs Attention” will be a thing of the past 👍

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg