prusa3d/Prusa-Firmware #1846: MMU2S Singlemode uses filament in current pos, instead of asking

RakNoel, 2019-05-18:
Upon starting a new print in singlemode with an MMU2S it will sometimes just use the position the idler is in and load that filament instead of running Tx command asking for which filament to use.

Setup:
Octoprint, v1.3.11
MK3S, v3.7.1
MMU2S, v3.7.1

bynnek, 2019-05-18:
When octoprint connects to the printer, the printer essentially resets and the MMU2s has to reinitialize.  This takes like 20 seconds or so.  If you start a print with octoprint before the MMU2s has initialized, then you will get the error you were experiencing because the MMU2s isn't ready.  Best way to avoid it is after you connect with octoprint, wait till you hear the MMU2s re-home the drum before you submit a print job.

RakNoel, 2019-05-18:
Would it then be possible to have printerstatus busy while waiting for MMU2s calibration?

mkbel, 2019-05-18:
This delay is unnecessary. We will fix it. http://dev.prusa3d.com/browse/PFW-872

closed 2023-09-28 by github-actions[bot]