prusa3d/Prusa-Firmware #1849: Elapsed time is much too high after print

PxT, 2019-05-20:
See photo at https://i.imgur.com/XUooqti.jpg.  
This print took about 18 hours, but printer thinks it was 1143 hours.
Printer: MK3/S + MMU2/S. 

Gcode is the MMU2S version from https://www.prusaprinters.org/prints/168-universal-auto-rewind-spool-holder-mm/files

PatitoPatozo, 2019-06-30:
Have seen this too with the standard 2-color Benchy. "1170 hours 30 minute"

Print was successful (with a few loading issues - that were easy to solve).

[image]

fdurrer, 2019-07-11:
I've observed this too on certain MMU prints with user interaction. As it happens only occasionally I think it depends in the type of user interaction but I have to investigate further when it happens next time.

jsilva-iii, 2019-07-20:
I have a Mk2.5S+MMU2S  and have observed exactly the same - I have 10-20 Load failures and 2 MMU failures
[image]

jsilva-iii, 2019-07-21:
I have just looked at my overall stats and that is also messed up 
[image]

jsilva-iii, 2019-07-21:
I have a MK2.5s + MMU2S FW 3.72 -  this is a new Rambo as of 24 June.

uniacke1, 2019-08-08:
Ditto, I've observed it when prints have been paused in some form or another several times, manual pause included. I'd paused it to inspect if a layer shift was recoverable.

vintagepc, 2019-08-17:
Still present in 3.8.0-RC1. Print was interrupted/paused by "MMU needs attention" and is actually ~6h 

[image]

Kadah, 2019-08-20:
Looks like an integer underflow.

PR #2309 (2019-12-15): Fix MMU needs user attention Statistics error
If you encounter a "MMU needs user attention" error, fix the issue and let the print finish, then the total print time in Statistics will show bogus numbers. This PR prevents this from happening.
Closes #1849 
Closes #1922 

PFW-1070
closed 2019-12-15 by DRracer