prusa3d/Prusa-Firmware #1851: 3.7.1 Firmware for MMU2/MK3S Single Extruder Print not purging filament anymore

TimothyGold, 2019-05-20:
I  have upgraded both MK3S firmware to 3.7.1 and am also using the new Prusa Slicer RC and now MMU2 Single filament prints no longer print a purge line and will continue the print missing part of the first layer.  I don't know for sure if this a Slicer issue or firmware issue.  I checked the start G-Code in the slicer and it is the same as the old slicer so I don't know why it is no longer working.  This happens even if I load filament to nozzle.  Just doesn't push enough filament to the nozzle.

RakNoel, 2019-05-22:
Tested now with both octoprint and print-from-sd in singlemode, this is not an issue I experience.
FW 3.7.1
MK3S MMU2S
Octoprint 1.3.11
PrusaSlicer 2.0.0 + win64

Printer purges dropp, skips about 3cm then purges to end and back.

TimothyGold, 2019-05-22:
The only thing I changed in the config after updating to 3.7.1 was to enable the filament ir sensor in all prints.  The sensor is obviously used in all MMu2 prints.  I will try a print tonight with this disabled on a single print to see if I can get it to act like the previous firmware.  Not sure what has changed other than this....

RakNoel, 2019-05-22:
Ran tests with both IR and F.I.N.D.A working, both crash detection and filament-detection enabled.

TimothyGold, 2019-05-22:
I do have a little longer PTFE tube from MMU2 to the extruder.  Do I need to re-calibrate this?  I thought that this didn't matter anymore.  I tried 5 different prints last night trying different methods loading first to the nozzle and then letting it load from mmu2.  The filament looks like it loads and I can see the IR Sensor sense the filament going to 1 then it puts the filament in and then retracts it a little just like it did in the old firmware but when it starts the purge the lines don't show up and the border circle does not show up and only starts to purge filament after it is in the print.  I guess I will also try to downgrade tonight as well to see if i am successful again...

gudnimg, 2023-01-30:
Adding this to thr 3.13 milestone as a reminder to close this issue.

closed 2023-07-31 by gudnimg