prusa3d/Prusa-Firmware #1852: 3.7.1 Filament change request during print

mvasilakis, 2019-05-20:
I've had an odd issue a few times today where I got a request to change filament during a print. I checked the slice and I didn't accidentally add a filament change during the print. 

On [facebook several others have reported similar oddities with respect to filament change](https://www.facebook.com/groups/prusacommunity/permalink/1001169140223818/). I cannot check the IR sensor during the change request because the menu locks me until the filament change is completed. I did continue the print with the screen set to show the IR sensor state and it stays steady at 1. 

MK3s MMU2s firmware 3.7.1 (final not a RC) running as a MK3s with the MMU2s cable disconnected. 
Model sliced with Prusa Slicer 2.0.0 rc
Printed via USB - latest stable Octoprint

WarrenSchultz, 2019-05-20:
Ran into this issue printing from SD, using the MK3S/MMU2S, after running out of filament on a single-filament print. (Single extruder gcode.)

I can watch the terminal via com if there's a way to enable that for debugging during the print. As mentioned in the report above, I can't see the sensor status during the error via the screen.

Smithc9494, 2019-05-20:
Ive had the same issue printing from both SD and Octoprint. My filament sensor it turned off and at several points through the print it wanted to do a filament change for no reason. I looked at the Gcode for a M600 color change and didn’t find anything. I’m on the most recent firmware release.

michalxfanta, 2019-05-21:
Could you send us a screenshot of LCD with printer version after printer restart, please?

mvasilakis, 2019-05-21:
> Could you send us a screenshot of LCD with printer version after printer restart, please?

Yes. The printer is in the midst of a long print. I'll copy a link to the photo(s) once the print completes.

teookie, 2019-05-21:
This has happened to me as well printing a large file from SD card. No M600 was found in the gcode. 3.7.0-2201 firmware downloaded direct from prusa.

RakNoel, 2019-05-22:
# Having the same issue
Im having the exact same issue. Printer stopping randomly during print and ask to change filament. Beeping loudly. Happens with both Octoprint and SD. Usually a coupple of hours betweeen each time.

## Debugging
 * Have tried moving the IR sensor closer, reads normaly
 * Have pressed F.I.N.D.A a bit down
 * Tested both Octoprint and SD

## Setup
FW: 3.7.1
HW: MK3S, MMU2S
Slicer: Prusa Slicer 2.0.0

##Attatchements
Only first 5% due to copyright. Happend 2 times within this span.
[the_model_used.gcode.txt](https://github.com/prusa3d/Prusa-Firmware/files/3208001/the_model_used.gcode.txt)

mvasilakis, 2019-05-22:
> Could you send us a screenshot of LCD with printer version after printer restart, please?
Here you go:
https://www.dropbox.com/s/2i6ghes6md1eeme/IMG_5817.JPG?dl=0
https://www.dropbox.com/s/45841kl4lf4713i/IMG_5816.JPG?dl=0
https://www.dropbox.com/s/sn3mu11d8auvwxj/IMG_5814.JPG?dl=0

WarrenSchultz, 2019-06-04:
Out of curiosity, for the others having this issue, are you using Prusa-printed or Prusa-PETG parts?
I'm using Atomic CFPETG, and I'm wondering if there's some light bounce going on, or just a coincidence.

mvasilakis, 2019-06-04:
I went almost 12 days without this issue and today I had it again, twice so far. I burned almost a whole spool of Hatchbox silver without issue (after the 1 or 2 that I initially had). A couple days ago I put in a spool of prusament and ran 3 - 8 hour prints without issue. Today On a small print it happened twice. At a similar spot in the model, oddly enough. 

My printer is in an enclosed environment. LED's light strips in a NewAge garage cabinet. The room is air conditioned. 

In my case I reprinted the idler door with matterhackers black ABS since the prusa supplied one would not trigger the IR sensor. I did not have the issue prior to 3.7.1. But that's not a fair assessment since I only had the printer in it's current configuration a week or two and I was battling with the MMU2s during that time where I faced several filament issues because of the supplied extruder idler door & IR sensor.

ravender83, 2019-06-13:
Same by me. I've upgraded to 3.7.1 and at start it went to change filament. Before 3.7.1 I was using 3.5.x and didn't had such problems.

theaquarium, 2019-07-15:
Still an issue in 3.7.2

jennech, 2019-07-21:
Same issue with 3.7.2 and black Petg mk3

darrenph1, 2019-09-24:
I am getting this now on 3.7.1  Did anyone figure out a workaround?

mvasilakis, 2019-09-24:
> I am getting this now on 3.7.1 Did anyone figure out a workaround?

I haven't had it for a few weeks now. The only change I made was switching off crash detection for another reason. Maybe give that a shot and see if it yields an improvement. Admittedly I can't think of one reason that this should help.

theaquarium, 2019-09-25:
I'm not really sure if this is a fixable issue or whether it even is an issue with firmware. I have since upgraded to the MK3S with the new filament sensor, and it has been working fine for me. If you can't upgrade, I would try cleaning the optical filament sensor lens, as I noticed that mine was grimy and dusty when I upgraded. As a last resort, you can always just disable the filament sensor.

darrenph1, 2019-09-25:
This might have been caused by me moving back to PrusaSlicer 2.0 from 2.1. It was slicing an openScad file with some gaps in the wall. Someone pointed me today to autodesk netfab site and it fixed the stl. So I sliced it again in 2.1 and it looks good. I won't be able to start this until the morning but I will report back how it goes.

RakNoel, 2019-09-25:
Have experienced this from time to time, but it went crazy on a octoprint plugin. Disabled and re-uploaded gcode which fixed issue

mvasilakis, 2019-09-25:
> Have experienced this from time to time, but it went crazy on a octoprint plugin. Disabled and re-uploaded gcode which fixed issue

Can you share which Octoprint plugin so we can test it?

darrenph1, 2019-09-26:
Just wanted to post an update. I am 10 min away from finishing an all day print that I was unable to do without the filament change popping up for the past few days. What changed: I ran the STL through netfab and then sliced it in slicer 2.1 (previously I had tried going back down to 2.0 which I think caused part of the problem)

closed 2023-09-28 by github-actions[bot]