prusa3d/Prusa-Firmware #1853: FW 3.7.1 - PSU overheating after about 5 minutes printtime with PETG

kgolger, 2019-05-21:
I use a Cosel power supply PLA300F-24 (24VDC, 300W). After installing firmware 3.7.1, the power supply switches off after about 5 minutes due to overheating. After cooling down, I restart the printer and the error repeats itself. After downgrade to firmware version 3.7.0, everything runs normally again. What was changed in 3.7.1?

Hardware:
-- original Prusa i3 upgrade MK3S with MMU2S
-- octoprint 1.3.10 with octopi 0.15.1
-- image:
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
-- raspberry 3B+
-- slic3r 1.41.3 64-bit PE

michalxfanta, 2019-05-21:
In the firmware version 3.5.2 and higher, we have increased the PWM frequency to reduce noise from the new PSU (black case). In some cases, the Rambo board overheated. We have now reduced this frequency to 32 Hz so that both, old PSU (silver case) and new PSU (black case) are quiet enough and not overheating. But you have non-original Cosel PSU.

kgolger, 2019-05-21:
Thanks for the quick reply. I am using an original Cosel power supply PLA300F-24 from mouser.com (https://www.mouser.at/ProductDetail/Cosel/PLA300F-24) because the original Prusa PSU was gone. You have reduced the frequency to 32 Hz. What value did you use before? I would like to change this in the source code and recompile 3.7.1. It's worth a try.

michalxfanta, 2019-05-21:
The original frequency was 8 Hz.

Panayiotis-git, 2019-05-21:
The PSU should not overheat with either 8 or 32 Hz.

kgolger, 2019-05-22:
"_We have now reduced this frequency to 32 Hz_..".
**reduced** from 8 to 32 Hz (Hertz)?
I will give it a try, but i think there is another reason.
With FW 3.7.0 my printer works over 10 hours without any kind of problems, no overheating, no MMU problems...

leptun, 2019-05-22:
The frequency was initially 8Hz, it was increased to ~42kHz to combat noise, but resulted in overheating boards. Recently it was reduced to 32Hz to address the overheating issue

ochm, 2019-06-01:
The bed power switch frequency is very anoying and noisy, the sound goes from bed, and is amplified by metal sheet, more amplified with powder coated PEI. 3.7.0 was OK.

kgolger, 2019-06-15:
I replaced the PSU. With the new Meanwell PSU-500-24 now if have no problems with V3.7.1, V3.7.2RC1.
Thanks to all for the support!

closed 2019-06-15 by kgolger