prusa3d/Prusa-Firmware #1857: feature request: automatically disable stealth mode during mesh bed leveling

teookie, 2019-05-22:
I think this should be easy for someone who knows their way around the firmware (definitely not me).

I always print in stealth mode, but mesh bed leveling takes approx 20 seconds longer than normal mode. Based on the sounds the printer makes and nothing else, it appears to turn off stealth mode for the z moves, and turns it back on for the x and y moves. All this switching on and off  adds ~20 seconds to the mesh leveling process.

Normal mode: ~43 seconds
Stealth mode: ~63 seconds

So, it would be great if stealth mode was automatically disabled for the entire mesh leveling process.

cimoalpacino, 2019-05-22:
Which fw version are you using? The new mesh bed leveling (fw > 3.7.0) is very fast

teookie, 2019-05-24:
It's very fast in normal mode. But in stealth mode it's about 20 seconds longer. My request is that the printer automatically switch to normal mode prior to performing bed level.

vintagepc, 2023-09-13:
This can actually now be done by using the M914 and M915 g-codes in your custom printer start code before and after the G80 call. 

Example snippet I use myself:

```
G28 W ; home all without mesh bed level
M914 ; High power MBL
G80 ; mesh bed leveling
M914 R ; Resume previously set mode for print
```

Closing this since it's been implemented.

closed 2023-09-13 by vintagepc