prusa3d/Prusa-Firmware #1860: Half of print worked well, but then the print shut down when color change was needed. Why?

ErrorlessJinX, 2019-05-22:
Currently, I'm working in an internship where I'm working heavily with Prusa 3D printers. The primary printers I've been working with so far is the Prusa i3 MK3S. I've been slowly getting some things dialed in on it, but had an issue come up that no one seems to understand why it happened.

The first 3/4 of the print went great. However, there was a color change needed around that point but I had been in a meeting for a while during that time. The printer head had moved to its normal position for the material change, but when I got to the printer, the head and pad had both completely cooled down and the printer didn't show as having any print in the queue. Will the print time-out if the material is not changed soon enough? Or is there something else going on that I can't try to fix?

I'd like to fix this issue because my prints are in the range of 24 to 28 hours for each print. It would be nice to not have the prints fail after getting 20 hours into a print. I will submit any info if you need it, but I'm assuming it is just something with the printer overall because it has done fine with plenty of other prints using the same settings, material, etc.

ErrorlessJinX, 2019-05-22:
I'm currently unaware of the firmware version because the printer is at my work but I can check when I go back in tomorrow.

I was printing from an SD card. There were no power outages of any kind at any time during the print.

leptun, 2019-05-23:
Could you also post the gcode you were using? Maybe I can replicate the problem.

scalda, 2019-11-20:
Hi, I have been having this issue over this last week as well. 
my Prusa would either park for a genuine filament change as it has run out or a false positive due to the optical sensor on the MK3 and if it's left for too long then it shuts off the heatbed and I get an errtemp message on the screen this seems to happen after the nozzle has cooled off like it is supposed to when a filament swap is waiting too long.

I am running the very latest MK3 firmware 3.8.1 OCT 24th 2019 version
This has happened on 4 different gcodes
and has happened 4 times now all at varying times of day/night

closed 2023-09-28 by github-actions[bot]