prusa3d/Prusa-Firmware #1862: 3.7.1 MK2.5 heatbed clicking on heat

magnazan, 2019-05-23:
Updating to the 3.7.1 firmware for the MK2.5 I have two machines experiencing the same issue where the heatbed is making a loud fast clicking sound from where the connectors/LED are. Rolling back to 3.7.0 and both machines are quiet again. Connectors have been double checked and they are correct as per the manual.

jurcja1, 2019-05-23:
I agree with this issue, facing the same after FW upgrade to 3.7.1.

F0x06, 2019-05-24:
Clicking issues are back due to the heatbed transistor overheating issue, prusa reduced the frequency a bit, because high frequency is quiter but generates too much heat and can damage the printer heatbed transistor.  See: https://github.com/prusa3d/Prusa-Firmware/issues/1636

Dippyskoodlez, 2019-05-25:
Bed and PSU clicking here, US 120v/60Hz power, MK3s on 3.7.1.

skonakov-tophat, 2019-05-27:
Having the same rapid clicking noise with 3.7.1 on MK3S :(

devcodeio, 2019-05-27:
Likewise, PSU is clicking constantly once heatbed hits the preheat temp.  I'm in Canada so 120v/60Hz power as well.
My LCD is also flickering a lot now.   Probably should have its own bug logged?
None of this happened with the previous firmware.   Id rather add a fan to the PSU to help with heating then have it click.

leptun, 2019-05-27:
@devcodeio The electronics board is heating, not the power supply. The clicking is probably caused by something related to using 110V/60Hz. In the Czech republic the mains power is 220V/50Hz. This is not something they could have anticipated would hapen on that voltage since the testing is done in house.

I am also in europe and I only had clicking bed with the first switching speed (8Hz). It is much better for me now with 32Hz. Power supply is external and quiet (and original).

devcodeio, 2019-05-27:
@leptun Hmm, The heatbed kinda has a buzz to it, but the PSU itself actually clicks.  It's also more apparent at higher temps.

leptun, 2019-05-27:
I still think this is caused by using 110V instead of 220V. The real problem is that fixing this through software is close to impossible because of hardware limitations.

To fix it you would need a higher frequency on the bed, but that causes overheating on the main board. Issue: #1636

Lowering the frequency back to something more reasonable fixed the issue mentioned above, but reintroduced the clicking. I for one am very pleased of the current state of the clicking, but others have problems. It is to be seen how prusa will address this issue.

Btw, what psu version do you have? (The silver one or the black one?)

devcodeio, 2019-05-27:
I have the silver model PSU, which from my understanding is an older model. Ive read that when the issue first became apparent a year or two(?) ago that some users received a replacement from Prusa. Although im not sure if it resolved their issues. Something i'll have to look into. 
I'm ok with the clicking if i know with some certainty that it makes for a safe and more reliable system.  But hearing unusual sounds like this using usually instills a bit of worry in most people, and rightly so.
@leptun  Thanks for the info,

magnazan, 2019-05-27:
Just a note I'm on 230v 50HZ here in New Zealand. I have two MK2.5 doing exactly the same thing.

tkircher, 2019-05-31:
I have an original MK2 that I upgraded to a MK2.5S, currently on FW 3.7.1. The heat bed and power supply are dead silent when everything is warming up. The only thing I can really hear is the faint breeze of the Noctua fan.

devcodeio, 2019-05-31:
@tkircher The issue should not be apparent while heating up, it's only after the required temps been reached and its trying to keep the temp stable.
At least that's been my experience so far, also it doesnt seem to happen for everyone.

tkircher, 2019-05-31:
@devcodeio I tried it again, and everything is silent. Tried both 210/60 and 250/90, let it sit for twenty minutes at those temperatures. Monitored input power and bed voltage too. The problem might be that people don't have their connectors seated properly. I've seen some complaints of MOSFET burnouts and heatbed wire insulation and connector melting on both ends, and that's always caused by user error.

vertigo235, 2019-05-31:
This is normal, you guys must have new pritners and never experienced FW prior to 3.5.2.

vertigo235, 2019-05-31:
Prusa fixed the clicking by switching to a higher pwm frequency, but this caused other issues so they reverted back to a slower pwm and the clicking is back.  It's not as bad though.

tkircher, 2019-05-31:
I ran the 3.1.0 firmware on my MK2 (still the latest version) from the day it was released until I recently upgraded to the MK2.5S. I've never had any noise from my power supply. I'm aware that some people had this issue:

https://twitter.com/MoeJike/status/946795991004516352?s=20

It seemed that at worst it was a faulty power supply. Josef acknowledged it but this definitely does not happen on my printer.

sopis, 2019-06-03:
same rapid clicking noise on MK2.5S..I upgraded from MK2S, with Silver PSU...this never happened on my MK2S

Panayiotis-git, 2019-06-03:
The heated bed of the MK2.5, like the MK3's, introduces magnets to hold the metal sheet. These magnets interact with the pulsating current of the bed and acts as a loudspeaker. It depends on your specific bed if you are going to hear the clicking. If a magnet is loose or if you raise a little a corner of the sheet, the clicking becomes louder.

ochm, 2019-06-04:
The same clicking from heatbed on MK3, amplified by PEI powder coated steel sheet.

jiri-jirus, 2019-06-13:
yeah, so MK3 has hidden feature: integrated speaker in heatbed!!!

jiri-jirus, 2019-06-13:
it is clicking on my MK3 on 230V / 60HZ. and it does not matter whether MK3 is plugged into wall directly or into APC UPS.

cimoalpacino, 2019-06-14:
Jiri-jirus, you mean 230V/50Hz?😋

jiri-jirus, 2019-06-14:
yeah, i do. u guys confused me

ochm, 2019-06-15:
It is not a problem with power supply (wall outlet), it is a problem of firmware - driving frequency of heat power switch. The sound is pure physic - combination of magnnets in bed and heating spiral in bed - speaker.

DRracer, 2019-06-25:
As some of you have noticed, the audible sizzle is coming from the bed, it is not an electrical noise. The bed behaves like a loudspeaker and the 30Hz PWM responsible for switching heating on and off makes this sizzle. Right now we are working on a silent solution to this problem, we hope to have this fixed in FW 3.8.0.

probize, 2019-07-02:
Upgraded to 2.7.2 from 2.7.1 (prusa stock). Running 240/50Hz. Clicking noise seems to have incremented between the two releases. I have checked that connection is okay and have not been able to see any problems. Awaiting 3.8

Dippyskoodlez, 2019-08-08:
Just installed 3.8.0RC1 and not only is the PSU noise still there but it's far worse than before but at least it's only from the PSU now. The clicking is also causing my LED bench lamp to flicker quite aggressively too, and this is on an AC circuit I've verified is run correctly and it's on an APC UPS.

MakeSomeGood, 2019-08-08:
Can confirm what @Dippyskoodlez said. Heatbed doesn't have the rapid 'taser' sound anymore but PSU is clicking a lot more on FW 3.8.0RC1. I'm using a MK3S with a silver PSU in the US.

leptun, 2019-08-09:
@Dippyskoodlez Can you please post a video with the noise and flickering?

DRracer, 2019-08-09:
@Dippyskoodlez @MakeSomeGood thanks for your reports. If I understood correctly:
1. the bed noise is gone, correct?
2. the PSU is now making more noise that in FW 3.7.2? 110V/60Hz?
3. I don't understand the flickering issue, a video may help (but keep in mind videos are usualy 25-50fps, so not every flickering can be captured well). Could you please describe the issue in more details?

leptun, 2019-08-09:
@Dippyskoodlez @MakeSomeGood 

Are you confortable compiling the firmware on your own? If you don't, I can compile it for you.
I am suggesting you try the following thing: Change `HEATER_BED_SOFT_PWM_BITS `in Configuration.h from 5 to 7 and report back with the result. On my MK3 with silver PSU and 220V/50Hz I have no problem with or without this change.

DRracer, 2019-08-09:
@leptun are you sure the new finite automaton on ISR(timer0_ovf) respects this define? It may indeed... I haven't thought about it this way

leptun, 2019-08-09:
@DRracer Wait a second, I'm a bit confused. I didn't realize this define is no longer used. Oops.
@Dippyskoodlez @MakeSomeGood Nevermind.

Dippyskoodlez, 2019-08-09:
@leptun @DRracer 

Yes, the bed created noise I had previously has gone away. The PSU clicking however, went from an intermittent click to a much more consistent and rapid clicking. Here's a video of it on 3.8.0RC1 now:

https://youtu.be/M9iytZlMyEc

When working to isolate this for this video I found that heating ONLY the bed left be with an infrequent click while sustaining bed temperature. It was only when my hotend was brought to temperature, around 200-205C of the 215 it was heating to that it began to click. When the machine is up to temperature and idle, it's an intermittent click but nothing nearly as bad as the video above.

Once I start a print and motors begin, the clicking starts going wild like the video above.

- Printer is in stealth for noise isolation in video, clicking is present in both modes.

120V/60HZ us power, ~118v @ outlet according to my fluke, a spot on 60hz frequency according to my fluke and APC UPS.

It's difficult to catch my LED desk lamp flicker on my camera, I think the flicker has been there the entire time it's just a much stronger and very noticeable flicker with the new frequencies. It is on the same circuit as the Prusa PSU.

MakeSomeGood, 2019-08-09:
@DRracer 

> * the bed noise is gone, correct?

Yes, this is correct.

> * the PSU is now making more noise that in FW 3.7.2? 110V/60Hz?

This seems to be the case, the clicks within the PSU are more frequent in 3.8.0RC1 than in FW 3.7.2. 120V/60Hz in my case within the US.

> * I don't understand the flickering issue, a video may help (but keep in mind videos are usualy 25-50fps, so not every flickering can be captured well). Could you please describe the issue in more details?

This happens on my LED floor lamp. I tired to capture the light flickering on multiple cameras with different FPS but can't seem to really capture the flickering well. It looks like the loses some illumination for a tenth of a second coinciding with the clicks of the PSU. My lamp is plugged into another outlet than the MK3s but on the same circuit. There has also been other reports of the clicking and flickering on the Prusa3d Discord. 

To reproduce the clicking/flickering all I had to do was heat the bed up to 60c and when it reaches within ~1-2c of the set temperature it would start ticking. The clicking becomes more frequent if a higher bed temp is used (e.g. 70c or 80c).

Thanks!

vintagepc, 2019-08-09:
I'd like to update this issue for 3.8.0-RC1 - despite the changes, my silver PSU also still clicks quite noticeably. (Mk3S)

DRracer, 2019-08-22:
@vintagepc @MakeSomeGood we may have found the cause, the solution will be in 3.8 RC2, please stay tuned.

vintagepc, 2019-08-22:
Awesome!

vintagepc, 2019-08-22:
I see it was merged. I'll synchronize, build and test.

vintagepc, 2019-08-23:
I just took it for a spin. There's still the faintest of fluttering noises from the PSU, but it is **drastically** improved over both what it was before 3.5 and after 3.7.1, and you _really_ have to be listening for it.

DRracer, 2019-08-23:
@vintagepc thank you for confirmation, I'm very happy the new version (almost) solves the problem. The trouble with all PSUs is the fact, that they are internally switching between low and high power - that clicks. We must keep them in one of these states as long as we can.

vintagepc, 2019-08-23:
It's a little hard to tell but it may very well just be resonance from the heat bed itself as well (speaker effect.) But given it's mechanical I know you can't get rid of it entirely in software, only minimize it, and this does just that. :+1:

FloatingCameras, 2019-08-29:
In the US: Warmed up the bed and could hear a slight ticking.  Updated to RC2 and the sound was reduced to a very quiet hiss I could only hear I put my ear next to the printer and everything was quiet.  It wasn't bad to start with I would say now it isn't noticeable. 👍

DRracer, 2019-08-30:
@MakeSomeGood  @FloatingCameras thank you for your confirmation of the fix.

closed 2023-07-28 by DRracer