prusa3d/Prusa-Firmware #1872: SL1 - web interface - values missing

mylife4aiurr, 2019-05-24:
Some values are "???" on web interface but are present/filled in on SL1's screen. So if in same room as sl1 ok, but remotely no go. 
remaining time:
estimated time:
printing time:

firmware 1.0.5

[image]

filipkotoucek, 2019-05-27:
Thanks for reaching out. We are working on new release which will fix this issue. It will be released this week. Anyway, this is not quite propriate place where to create issues for SL1. Please use this space https://gitlab.com/prusa3d/sl1/sl1/issues next time:) Thanks

mylife4aiurr, 2019-06-02:
Looking for new firmware??...... posted a new issue and my sl1 log to address listed above

filipkotoucek, 2019-06-19:
This issue was fixed in newer versions. At the moment 1.1.1 is current stable version. You can updat ether via net update or download update bundle file here: https://www.prusa3d.com/drivers/. @michalxfanta please close this issue.

closed 2022-05-23 by Prusa-Support