prusa3d/Prusa-Firmware #1873: Firmware update 3.7.1 broke filament loading

bcl, 2019-05-26:
I have a Prusa MK3S which has been working flawlessly since I built it last December, running Firmware 3.5.2

* Updated to 3.7.1
* Ran 1st layer test which worked fine, but it left the head low instead of raising it, which makes it hard to remove the plate without going into the z-axis adjust in settings.
* Ran a z-axis calibration, which failed. Not sure why.
* Ran the find home test and it worked fine, and was at the right Z
* pre-heated the bed and nozzle, had to raise it to grab the extra filament that always seems to drip out.
* Started a print of https://www.thingiverse.com/thing:2947391sliced by Prusa Slicer 2.0.0 (first time I've used the new version of the slicer)
* It immediately prompted to remove the filament,which wouldn't come out all the way.
* Futzed around and opened the access to the extruder. Filament was stuck, had a blob on the end (first time this has ever happened). Cleared it and removed the filament.
* Tried to re-load, it prompted me, inserted filament. No beep, no extruder movement.
* Moved extruder manually in the settings menu, works fine.
* Downgraded to 3.5.2 and it is printing just fine (after telling me there is a firmware update)

JohnOCFII, 2019-05-26:
The MK3S wasn't introduced until February of this year.  Perhaps you downloaded the MK3S firmware, when you really wanted the MK3 firmware.  They are different.  (Ask me how I know... -- but I caught my mistake before I started a print).

https://www.prusa3d.com/drivers/. (just scroll down a bit for the MK3 firmware)

bcl, 2019-05-26:
Ooooh. I had no idea there was a new model. Been too busy printing. Interesting that the flash utility didn't catch that and warn me. I'll give that a try after my current print finishes. Thanks!

bcl, 2019-05-27:
That was the problem, thanks for pointing that out!

closed 2019-05-27 by bcl