prusa3d/Prusa-Firmware #1874: FW 3.7.1 Temperature "Wobbling"

cycl0ne, 2019-05-26:
PRUSA i3 MK3s:

Problem with FW 3.7.1 the Temperature is going up and down up to 20degrees and wont stay at a fixed rate (example 220 degree, allways raising to 240 down to 190), this brings then the : thermal-runaway-error after some time.
In Octoprint you can see the spike of temperature very nice.

Downgraded to 3.7.0 and now all works perfectly again and i have a solid temperature line.

leptun, 2019-05-26:
Maybe some eeprom values got corrupted when upgrading. Try doing a factory reset as I have no problems running 3.7.1

cycl0ne, 2019-05-28:
Found the bug. it is the "quality" soldering of the thermal resistor from prusa. 1 week ago black cable came off.. now it was the red one. installed fw 3.7.1 and its stable.

closed 2019-05-28 by cycl0ne