prusa3d/Prusa-Firmware #1876: Feature Request: Software Disable MMU2

R0m3oF0x, 2019-05-27:
Please insert a **software setting** to **deactivate the MMU2**.

In my experience, the MMU2 cannot be used with abrasive filaments (stone and metal filaments) due to the long hoses. This only works if the filament is placed directly into the extruder (as it used to be), otherwise it gets jammed during printing.

Pulling the two plugs according to the MMU2 instructions is not a real solution in my opinion.

leptun, 2019-05-27:
Do you intend on having the ptfe tube disconnected or connected while having the mmu disabled?

I moddified the code to be able to disable it from menu, but did not do a pull request to the official firmware as I couldn’t test it and it would be kinda stupid just disabling the communication, but having the filament go through it.

If you want, I can port the changes to the latest version, but it will be english only.

R0m3oF0x, 2019-05-27:
Yes, I have the ptfe tube disconnected. Sounds just like I wished.

English shouldn't be a problem. Please port the changes.
Big THX!

leptun, 2019-05-28:
[FW371-Build2266-disable_MMU_from_lcd.zip](https://github.com/prusa3d/Prusa-Firmware/files/3227098/FW371-Build2266-disable_MMU_from_lcd.zip)

Here you go. Try it out an tell me how it works. The github project is https://github.com/leptun/Prusa-Firmware/tree/MK3_disable_MMU_from_lcd

One thing to note is that you need to reset the printer in order for it to have any effect.

mrcg, 2019-06-11:
I also would like to have feature to bypass the MMU2 for single printing.
It should also allow to leave the filament loaded in case of power off/on without the need to perform new load filament procedure when switching on. In about 95% of the cased i am printing with single filament and it's always a hassle to perform unload/load filament procedure when powering on the printer. Also in the forum lot of other user asking for single printing feature for the MMU2.

iamgnat, 2019-08-13:
I too would like to see this functionality be in the core firmware. As well as prints where it may be desirable to skip the MMU (due to filament or just avoiding extra complexity) it would also be an invaluable troubleshooting tool.

Sindarius, 2019-08-17:
This would be a great feature, I’d love to be able to do ninjaflex prints without going through the mmu.

leptun, 2019-08-17:
I might buy soon a MMU2S. If everything goes smoothly I plan to do some major changes to the communication layer between the printer and mmu that will also include this feature.

Zwirbel1, 2019-08-19:
Yes, please, this would be a great feature.

MadOtis, 2019-08-23:
I didn't search before I created a new feature request, so I'll delete mine.

I would like to see this, too, only with a slight twist... an unused gcode could be hooked in to disable the MMU, as well as or in addition to via the control panel.  I plan on designing a small "Y" connector that will sit between the bowden fitting at the print head that is threaded on one side (for the MMU's bowden fitting) and conical/funnel-shaped on the other input so there is no need to disconnect the MMU's PTFE tube each time I want to print flex, or the other abrasive filaments mentioned.

I suggest branching and creating a PR for others to approve and merge into master; I'm pretty sure there are more than 10 of us who would love/benefit from this feature.

maxcom99, 2019-09-18:
Is this going to be part of the final release so I don't have to use old firmware?

michalxfanta, 2019-10-11:
We are planning to add this feature to one of the next releases.

gitsh01, 2019-11-01:
Do you know which release? 

I want to use this feature when I'm printing quick makerbox samples. Running eight small 50g samples all the way through the PTFE tubes to the extruder is a pain.

maxcom99, 2019-12-20:
See no movement and this was closed for some reason...

maxcom99, 2019-12-20:
@michalxfanta please reopen

zapbang, 2020-02-03:
I'm with everybody else.  This would be an EXTREMELY useful feature for numerous reasons.  I AM a fan of the MMU2s and have had very few problems, but when printing the odd prints with FlexFill 98a or fillaflex 40d I would love to just bypass the MMU altogether and load up the filament straight into extruder.  

A menu option and/or a GCODE option would be perfect.

Please reopen or provide status.

Thanks!!

eblot, 2020-02-21:
TL;DR: I was about to create a new feature request, +1 for me

**Printer type** - MK3S
**MMU Upgrade** - MMU2S

**Is your feature request related to a problem? Please describe.**
  When printing with filaments that are not compatible with MMU2, MMU2 needs to be bypassed, directly filling the hot end with the filament as a MK3S standalone, not going through the MMU2. This causes several issues:

- PrusaSlicer refuses to slice for flex filament if a MMU2 is selected (the flex filament type does not even appear in the filament list), which means a standalone MK3S printer needs to be selected - this is ok and expected, but G-code gets generated for a MMU2-less printer, so each time the G-code is printed, the user needs to bypass the warning message.
- MK3S menus are different when a MMU2 is selected. For example, "Filament autoload / Load filament" is replaced with "Load filament" (for MMU2) and "Load to Nozzle", which also triggers MMU2 motors. 
- with the latest MK3S FW and MMU2 FW, I've experienced several times a weird issue:
  1. Print from SD, select a file
  2. Acknowledge the warning because the wrong printer is used (MK3S standalone G-Code on an MMU2-enabled MK3S printer)
  3. (Wait for preset temperature)
  4. Calibration runs (49 points)
  5. Hot end homes to 0,0 then
  6. NOTHING: no error message, no movement, and target temperatures fallback to 0. Nozzle is kept at Z ~= 0, ready to print the first extrusion line, but nothing occurs. That's it

I had this issue twice is a row (power cycling the whole printer each time). The third time I unplugged the MMU2 data cable, and everything went back to normal (no print issue, MK3S-style menu, no warning, perfect !)
  
**Describe the solution you'd like**
  It would be nice to add a menu to enable/disable the MMU2 so that on next boot/reset, MMU2 is fully ignored and MK3S behaves exactly as a standalone one: no more warning message with sliced code if the proper MMU2 option is selected, MK3S or MK3S+MMU2 menu available, no weird issue, no MMU2 unexpected movement after a printing error with a MK3S-standalone G-code, etc.

eblot, 2020-02-21:
BTW, why is this ticket closed till the feature is actually implemented?

leptun, 2020-02-21:
I think it was closed because it was added to the internal issue tracker. AFAIK the changes should come to 3.10.
3.9 is already packed with too many changes.

maxcom99, 2020-02-21:
@michalxfanta or someone from Prusa - please reopen / set expectations / clarify.

michalxfanta, 2020-02-27:
We planned this feature by 3.10.0

tareko, 2020-03-19:
Knowing this is coming soon prevented me from going in and modifying the hardware. Thank you!

diomark, 2020-12-06:
any updates on this? I spent the morning fighting my printer attempting to print TPE - mmu2s just doesn't work for certain filaments.  (I was attempting to feed it directly/fake it out, but it wasn't having it; normally this method works fine for other filaments but not tpe..)

alak, 2020-12-14:
Hey, any news? I need this feature too :)

moclub4, 2020-12-23:
Would be very useful! Any news here?

alak, 2020-12-26:
@michalxfanta is this still on feature list for the 3.10.0, and  if yes, when the 3.10.0 is planned to release ?

funkyskull, 2021-01-06:
This feature would be greatly appreciated. 
Reasons:
1. Trouble-shooting MMU2s issues would be much easier with the ability to quickly take the majority of the MMU2s hardware out of the equation.
2. In my experience, the MMU2s does not do well printing abrasives or flexibles. Having the ability to quickly switch to direct feed would be great.
3. This seems like really 'low hanging fruit' when it comes cost/benefit.

robscar, 2021-01-06:
+1, please add this feature

HarrySatt, 2021-01-07:
Yes. Please.

regards, harald

chrysos-adonis, 2021-01-09:
Please do the Software Disable MMU2 
Please.
I'd like to use my MK3s as before by software disabling the MMU2s, leaving autoLoading and autoUnloading the filament as before the upgrade.

Thanks a lot
Adonis

dukes4, 2021-03-06:
Iwould love this feature - any updates please

britxpatusa, 2021-03-15:
any updates on this? i see a feature request is in...does this mean we have to wait for a FW release?3.10 perhaps?
having to remove wires just to disable is crazy and then have to re-do calibration each time....i have no end of issues with MMU...keep going to single and get amazing results. past year...now im going back to MMU....but would like software option for single v MMU unload/load option....
thx

eblot, 2021-03-19:
It looks like it will not make it for [3.10](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.10.0-RC1), doesn't it?

boydbischke, 2021-03-24:
I too would like this feature.

JFKingsley, 2021-03-24:
Any updates on this? Would be extremely useful.

homan85, 2021-03-28:
Would love to see this function!

DRracer, 2021-04-14:
Dear community, we are aware of the various problems some users are reporting in relation the the MMU while others are very satisfied with the unit. We did a thorough research and most of the issues are a combination of more factors and the very raw user interface of the MMU itself.

After releasing FW 3.10, we'll do some serious improvement in the MMU firmware, especially in terms of user experience and reliability. We'll implement reasonable feature requests as well. The new version is expected during this summer (2021).

Also please note, some of the reported issues/requests are of hardware origin/limitation and probably cannot be solved purely by firmware. Yet it is too early to give an exact statement.

Thank you and happy printing.

moeilijk, 2021-06-20:
> 
> 
> We planned this feature by 3.10.0

3.10 doesn't seem to add this functionality yet. Was searching for this option as I have to print a broken part on the MMU2S that blocks me from using the MMU2S. Guess I still have to unplug it for now.

sboschert, 2021-06-24:
I would like to see this function, too!

5jvm0u4, 2022-01-01:
Same request, I need to bypass my MMU2S and the long PTFE tube because I'm printing very demanding flexible filament (75A), and the resistance is just too large with all that long PTFEs.
I've tried disconnecting MMU physically, but FINDA seems to be unhappy with that and pauses my print whenever I start it.

islander2013, 2022-01-16:
any news on this? we really need more support for the MMU2... we bought into it and now need this functionality.

homan85, 2022-01-17:
> any news on this? we really need more support for the MMU2S... we bought into it and now need this functionality.

I think MMU2 is end of life for Prusa now. They might give out a new firmware if there is a huge bug, but I think other than that we can count it as closed. 
I ended up doing the disable feature with hardware. It is not as good as software, but it is working. Can't count on Prusa anymore. We were just a bunch of BETA testers, and now the testing is done and Prusa has gone forward with other projects. 
That is at least my point of viewing it. 
I hope I am wrong, but the latest firmware update for the MMU2S is from 16/08/2019. That is 2½ years ago! I have received more updates on my (not smart) TV from 2018 during this time. It's a shame we wasted our money on the MMU2S. I wish I could turn back time and skipped it. My whole view on Prusa has changed since I bought the MMU2S upgrade.

edifabio-72, 2022-01-17:
> I think MMU2 is end of life for Prusa now. They might give out a new firmware if there is a huge bug, but I think other than that we can count it as closed. I ended up doing the disable feature with [...]

I totally share the thoughts of homan85.
All in all I managed to find a good compromise and good settings. I use MMU2 with satisfaction even with only one color.
I haven't done any hardware changes yet.
My only regret is the impossibility (perhaps again due to my inability) to print in TPU. This alone makes me want to do the hardware modification. I would have already made the change if there was no PINDA cable to cut.

DRracer, 2022-01-17:
I can say for sure the MMU is definitely not EOL, I'm looking at one running the new firmware right now. 
However, we are still working on the integration with the printer. 
I'm sorry the work has been delayed due to pushing the new products (XL) forward. Please hold on, the new MMU FW will come to the MK3/S/+.

edifabio-72, 2022-01-17:
> I can say for sure the MMU is definitely not EOL, I'm looking at one running the new firmware right now.
> However, we are still working on the integration with the printer.
> I'm sorry the work has [...]

This is great news.
I was convinced by now that after the release of the XL with more extruders, they would let the MMU2 project die

DRracer, 2022-01-17:
Nope, the XL definitely won't kill the MMU :wink:

eblot, 2022-01-17:
I've removed the MMU2 HW several months ago. Even w/ the same and single filament, it ended up blocking any new print.

Anyway, it is great news if a new FW is made available, but as long as there is now decent interface that comes with the MMU, every single trouble will keep being a nightmare to trouble shoot. It would be really nice if Prusa could release a new 32-bit board w/ a LCD screen able to drive both the MK3S and the MMU2 at once, with a common UI and most importantly a unified way to troubleshoot the problem - and ensure a better sync with both HW. I would have no problem to pay for such an upgrade.

leptun, 2022-02-19:
This feature, along with the new MMU2 implementation will come to fw 3.12. So far we’ve prepared the rest of the firmware:
- a lot of cleanup, ~6.5KB of flash saved, ~450B of ram saved. This was a much needed improvement overall and will be needed for the UI.
- Removed old, dead code that got in the way (eg SNMM, aka MMUv1)
- Some documentation in the xflash for the mmu2 fw update files.

The immediate plan is to isolate the existing MMU2 code with ifdefs, start removing what is not needed and prepare the filament sensor implementations (especially the pat9125 sensor from the MK3 and MK2.5) for the new code. This should happen the following week. After we confirm everything still works, we’ll begin adding stuff.

Because so many changes will be made in this version, the 3.12 fw release will most likely go through at least a beta and RC versions before the final release. We should have all the features ready by the first beta and then subsequent RC releases will have only bugfixes.

diomark, 2022-02-25:
Nice. very happy to see this get implemented..

diomark, 2022-11-15:
wow... I'm really surprised this is taking them so long to put in.. I'd use my printer alot more if this was there since I leave the MMU hooked up but usually don't want to deal with it..

leptun, 2022-11-15:
The 3.12 release had to be split in two unfortunately. As such the MMU changes were pushed to 3.13. This means however that 3.13 will release quicker than usual since it will only include the MMU as significant changes. The disable button is already implemented in the development branch (MK3), so this feature is technically done, but not useful yet since the MMU side of the firmware isn't released yet.

kalanchoej, 2023-02-10:
Hello from Feb 2023 and FW 3.12.1! Found this thread while attempting to print Fiberlogy FiberFlex and needing to bypass the MMU. Looks like I'll need to unplug some cables 😂 
+1 for this feature!

bmmiller, 2023-03-06:
Got my MMU2S about a week ago. Wondering if I should wait a bit? I know I'll appreciate being able to disable it easily when it's not needed.

3d-gussner, 2023-03-06:
@bmmiller Please try to use the MMU2 with firmware v1.0.6 to get a "feeling" for it. As already mentioned above we are working on new firmware for the MK3S + MMU which has a "software" disable function.

AJErazzor, 2023-03-22:
Pleeeeease add this to 3.13.0 - flex filament printer here and methods to trick the MK into thinking the mmu is loaded seems so silly.  Would rellay love to see this update out sooner than later this request is from 2019!

galop1337, 2023-04-02:
> Pleeeeease add this to 3.13.0 - flex filament printer here and methods to trick the MK into thinking the mmu is loaded seems so silly. Would rellay love to see this update out sooner than later this request is from 2019!

I agree with this. Right now I'm just unplugging the MMU2s and putting it beside the printer. An option to disable the MMU2 would be so helpful. It's been 4 years...

gudnimg, 2023-04-02:
In case it wasn't clear before, I just want to confirm that this will be included in the 3.13 release. It's already implemented.

galop1337, 2023-04-02:
> In case it wasn't clear before, I just want to confirm that this will be included in the 3.13 release. It's already implemented.

It wasn't clear to me, so thank you for the clarification. Much appreciated.

diablo1281, 2023-05-11:
The above mention of the added label `solved` is a bit confusing for me. Is there any update for MMU firmware?

Prusa-Support, 2023-06-10:
Sorry for the confusion.
The issue is solved but still open.

This is already fixed and implemented in the next firmware - already drafted but yet to be released.
Therefore the issue will remain open until the next firmware release.

Michele Moramarco
Prusa Research

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg