prusa3d/Prusa-Firmware #1884: Errors after  3.7.1  firmware upgrade

richelliot, 2019-05-30:
I upgraded the firmware on my i3 Mk3 to version 3.7.1 and have now encountered errors. 

Firstly, when I went to print the following error was thrown:

'some problem encountered z levelling enforced FIRMWARE 3.7.1'

I then started the collaboration wizard, which now gives me this error:

'selftest error please check filament sensor writing error'

I bought the printer assembled and have not had any problems with it until today.

leptun, 2019-05-30:
Did you by any chance upload the wrong hex file?

Eg: Mk3 on Mk3S or Mk3S on Mk3

richelliot, 2019-05-30:
Yes, it turns out I did!

I thought I had the Mk3 Model. I actually own the Mk3S.

Loaded the new Mk3S firmware and its now working again :)

Thanks for your suggestion.

leptun, 2019-05-30:
Glad to hear you have it fixed. You can close this thread now.

closed 2019-05-30 by richelliot