prusa3d/Prusa-Firmware #1895: XYZ Calibration stopped working in MK2.5S

marcopiraccini, 2019-06-05:
Hi, 

MK2.5S, just upgraded firmware to 3.7.1-2266
XYZ calibration now does not work anymore, but the real problem is that I don't see anything wrong in log file (attached). So:
- Please add more info in these logs (feature request)
- Why is not calibrating?

leptun, 2019-06-05:
The points look too much to the right, barely missed by the probe. Could you please make a video of it searching for a point?

marcopiraccini, 2019-06-09:
Any hint? The 4 calibration points are found. Distances seems OK...
It would be REALLY Good to have some hint from the firmware log (like: calibration failed because...). 
Now the only indication is "Consult the manual..."

marcopiraccini, 2019-06-09:
Unmounting and remounting x axis fixed this. So closing this issue. But I think that firmware should really be much more verbose on errors

closed 2019-06-09 by marcopiraccini