prusa3d/Prusa-Firmware #1898: M1 fails in many situations

GWdd, 2019-06-05:
Adding M1 to a gcode seems to almost never work; it is extremely position dependent; and having more than one can corrupt later insertions that will normally work.

For example this segment of code pauses, but displays nothing (filename is present)
```
M115 U3.7.1 ; tell printer latest fw version

G91
G0 X5 Y5 Z.1 F1000
M1 FIRST BASE
```

However this segment from a larger piece of code works (sometimes)
```
G0 X0 Y200 Z100 F3000 ; this is a good waiting point for bed flip

M1 Flip sheet then press OK 
```

And if the M1 First Base command is in the sequence, the second M1 fails to produce text.
```
M115 U3.7.1 ; tell printer latest fw version

G91
G0 X5 Y5 Z.1 F1000
M1 FIRST BASE

G90 ; use absolute coordinates

M83 ; extruder relative mode
G28 W ; home all without mesh bed level

; heat things up
M104 S150; [first_layer_temperature] ; set nozzle temp
M190 S40 ; [first_layer_bed_temperature] ; wait for bed temp
M109 S150 ; [first_layer_temperature] ; wait for extruder temp

G80 ; mesh bed leveling

G0 X0 Y200 Z100 F3000 ; this is a good waiting point for bed flip

M1 Flip sheet then press OK 
```

Not being able to produce meaningful text messages on the LCD prevents users creating useful scripts to handle odd print conditions.

This is similar to M117 defect, also where desired text is NOT being display when it should be.

I tried many variations, including add G0 commands prior to the M1 - in various places in the gcode, and honestly now believe it was simple chance I found a place where the command actually works, as shown here:
```
M115 U3.7.1 ; tell printer latest fw version

G90 ; use absolute coordinates

M83 ; extruder relative mode
G28 W ; home all without mesh bed level

; heat things up
M104 S150; [first_layer_temperature] ; set nozzle temp
M190 S40 ; [first_layer_bed_temperature] ; wait for bed temp
M109 S150 ; [first_layer_temperature] ; wait for extruder temp

G80 ; mesh bed leveling

G0 X0 Y200 Z100 F3000 ; this is a good waiting point for bed flip

M1 Flip sheet then press OK 
```

vojtech-pavlik, 2019-06-05:
I wrote a patch that fixes M0, M1 and M117 to work reliably. Hopefully that fixes it for you.
[https://twilight.ucw.cz/prusa/M0-v2.patch](https://twilight.ucw.cz/prusa/M0-v2.patch)
Now I need to figure out how to get it merged.

vojtech-pavlik, 2019-06-05:
Pull request #1900 created.

3d-gussner, 2021-02-18:
Should be fixed with PR #3034

closed 2023-09-28 by github-actions[bot]