prusa3d/Prusa-Firmware #1906: Thermal Runaway after Filament Runout on MMU2

bwright86, 2019-06-11:
**Printer:** Prusa i3 MK2.5S w/ MMU2
**Printer Firmware:** 3.7.0-2201
**MMU2 Firmware:** 1.0.5-297
**Octoprint Version:** Version 1.3.10

### Explanation:
Had an issue today where the filament was running out, and the very end was getting stuck in the MMU2 bondtech gears. I unscrewed the PTFE tube connector and pulled on the filament while a print job was starting. The extruder carriage stopped moving and the display screen stopped updating, Octoprint showed a "Communication Error". Eventually I smelled smoke coming from the extruder, and realized it was getting too hot, so i restarted the printer. After booting up, the display showed the extruder was at 548 degrees!!! It triggered a MAXTEMP error, and a long steady beep emitted from the printer, and both fans began to cool the extruder.

### Issue Theory:
An issue exists with filament runout in the MMU2 that causes an exception or panic in the RAMBO.

leptun, 2019-06-11:
This problem is very scary since if you weren't there it could have done irreversible damage.

I see that you are using firmware 3.7.0 on MK2.5S. In firmware 3.7.1 there was added a security feature that prevents this from happening on MK2.5(S) printers. The watchdog timer is used so that if the firmware freezes for any reason it just resets it and only results in a failed print. You should update the firmware to avoid a fire. As for what caused the firmware to hang, that is something to test for Prusa/people with the MMU2.

bwright86, 2019-06-12:
Thanks for the response, I did notice the firmware update was available, and it appears to include a WatchDog safety check in the update. I am a little surprised, as i believe the MK2S firmware had this feature when I originally purchased the printer.

It would be worth it, IMO, to alert MK2.5 owners to the feature update, to get them on a safer version. I feel like Prusa printers are safer printers normally, but this is because good quality parts and good safety features make them this way.

I plan on patching my printer shortly, and will report back and close this issue.

Thanks again for the quick response!

bwright86, 2019-06-14:
It appears that updating the firmware has resolved this issue.

I will mark it as closed.

closed 2019-06-14 by bwright86