prusa3d/Prusa-Firmware #1916: Feature request - purge after pause

ohthetrees, 2019-06-13:
A little molten filament inevitably dribbles out during a pause. This blob can be a big problem when resumed, and has caused me failed prints. I suggest that after a print is resumed, that another purge be performed.

ManuGithubSteam, 2019-07-07:
Hi

This can be problematic: Imagine you print something big on the build plate and the extruder can not reach down to the front anymore without touching the print. Like the cables your scrap over that part etc.

However, Purge into infill of the part would be possible i think but im not sure it is implemented yet for machines without mmu2s.

leptun, 2019-07-07:
Purging into infill is not possible since it is something the slicer decides, not the firmware.

ohthetrees, 2019-07-09:
Good point. But it could still purge into the air like it does with an M600 filament change.

michalxfanta, 2019-10-11:
Thank you for your interesting idea, we will add it to our backlog.

closed 2019-10-11 by michalxfanta