prusa3d/Prusa-Firmware #1917: Compile error: 'FARM_DEFAULT_SAFETYTIMER_TIME_ms' not declared

nebbywan, 2019-06-13:
With the change that ec2f194350adbd18df97bba8051992d15e6bca57 made to line 7712 of Marlin_main.cpp, I now get the following error when compiling the sketch: 

`Marlin_main.cpp:7712: error: 'FARM_DEFAULT_SAFETYTIMER_TIME_ms' was not declared in this scope`

vertigo235, 2019-06-24:
There were additional defines added to the variant files, you are probably using a customized one.  Add the new defines and it should resolve this.

vertigo235, 2019-06-24:
... your Configuration_prusa.h file, check the recent updates to see what new defines you must add

nebbywan, 2019-06-24:
@vertigo235 Yup, embarrassingly, that was it. My file was stale and apparently I wasn't paying close enough attention to the revision history to see the changes. Thanks!

closed 2019-06-24 by nebbywan