prusa3d/Prusa-Firmware #1923: MK3s/MMU2s - Repeated failures unloading filament after load fails

b13johnson, 2019-06-15:
Versions:
PrusaSlicer-2.0.0+win64-201905201652
Windows 10
Prusa MK3s with MMU2s running 3.7.1 (current release)

Behavior:

I have been having issues with multiple interventions required during a print. I have optimized the nozzle temperature for my setup and set PrusaSlicer to do two cooling moves when unloading filament. Everything is working better, but still not where I think it should be. I created a two color "torture" test for optimizing and evaluating performance of the system. It consists of two 50 mm tall calibration towers, which requires 333 tool changes to complete.

When the printer says that the MMU needs attention, it is almost always during an unload (LED's blinking rapidly), but it is while trying to load that same filament. It can't unload because the filament is ground down and the BondTech pulley in the MMU can not grab the filament. I noticed that the BondTech pulleys in the extruder are still engaged with the filament when I try to remove it to continue the print.

This seems like a possible firmware issue, but I can't figure out if the grinding is happening during the load or after the load fails and it tries to unload.

mwjones1971, 2019-07-02:
PrusaSlicer-2.0.0+win64-201905201652
Windows 10
Prusa MK3s firmware 3.7.1
MMU2s firmware 1.0.5

I encountered the same issue, but can expand on what is happening. I took an object I've printed before and downloaded an updated version and resliced it with PrusaSlicer (GCODE attached as ZIP file).

On certain tool changes, the extruder would complete the ramming and then back the filament out of the bondtech. The MMU Idler instead of immediately engaging would "re-home" (and the MMU LED's would flash as if the control board reset) and then try to unload. Because of the delay in unloading the tip becomes a small blob instead of the usual pointed tip, and thus would jam trying to unload. Only recourse is to remove the PTFE tube from the selector and the extruder, pull the ground down section out of the MMU and cut the 48cm of filament out. Once this is complete the resume button on the MMU would reset the Idler again, complete unloading the remaining short piece, move the selector to the next material and then reload... slowly (it appears to be trying to determine the length from MMU to Bondtech so it loads in steps of approximately 4cm until the IR is triggered by the bondtech).

In my scenario it was typically switching from Filament 2 to Filament 1 (and not consistently), but it did do it once or twice on the reverse. 

[v3.3_Rewinder_MM_0.2mm_PLA_MK3SMMU2S.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/3347785/v3.3_Rewinder_MM_0.2mm_PLA_MK3SMMU2S.gcode.zip)

angelshtilianov, 2019-08-27:
I've just assembled my MMU and I've hit the same issue. When changing filament it seems that immediately when the IR trigger the extruder motor stops immediately and the filament remains still engaged by the bondtec gears. Other than that the bontec gears start almost immediately after that rotating pushing to the extruder and MMU if succeeded to take off the filaent from bondtec is pulling the filament to itself.
Shouldn't the rotation directions be synchronized between the two motors?

RussPitcher, 2019-10-06:
I’m seeing very much the same issues. I’ve only had the printer for a month or two and I first built it as a standard MK3S and found it to be superb and very reliable. After a couple of weeks of getting used to it I added the MMU2S. Since then I’ve had a lot of problems, especially after cancelling a print, even when printing single colours. I tried my first multi filament model tonight and gave up after about 16 layers and 90 minutes of constant attention. I think I only got about 8 changes in one go without intervention.

b13johnson, 2019-10-26:
I took a break after doing the best I could do with the firmware available at the time and optimizing the hot-end temperature for multi-material printing. There was still too much hand holding with the many interventions required to successfully print. It was not fun, so I took a break for a while.

I'm happy to report that after downloading the latest software/firmware, PrusaSlicer 2.1.0, MK3S 3.8.0 and MMU2S 1.06, I've been having excellent success. I have also decided to switch to Matter Hackers Pro PLA, abandoning the remaining Hatchbox PLA I had laying around. I have printed several four and five color prints with little to no intervention. It seems like they have improved the firmware quite a bit.

It is important to get the hot-end temperature correct for the filament and thermistor being used in your setup, but otherwise things are looking pretty promising. One of the prints had was four colors and about 1000 tool changes and it printed without any errors. I'm impressed!

neilgupta, 2020-03-14:
Seeing the exact same issue with v3.8.0 on a new MMU2S with an MK3S. Commenting in the hope one day Prusa decides to fix this... until then, it's very disappointing that Prusa attaches their name to this thing.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg