prusa3d/Prusa-Firmware #1925: electric noise hot bed 3.7.2 rc1

austaquio32, 2019-06-17:
good morning, after updating to 3.7.2 rc1he noticed a substantial noise in the print base after warming up to maintain the temperature, I saw that the bed pwm was implemented, with the two printing bases makes me noise and not It is very nice, does this have a solution?

rlightner, 2019-06-20:
Noticing this as well.

DRracer, 2019-06-24:
If you mean the audible sizzle coming from the bed, it is not an electrical noise ;) . The bed behaves like a loudspeaker and the 30Hz PWM responsible for switching heating on and off makes this sizzle. Right now we are working on a silent solution to this problem, we hope to have this fixed in FW 3.8.0.

ochm, 2019-06-25:
Duplicate with #1862

closed 2023-09-28 by github-actions[bot]