prusa3d/Prusa-Firmware #1927: Feature Request: Starting Octoprint File from LCD

tbedore, 2019-06-17:
A feature that would make my life much easier.  The printer has the ability to see the files that are on the Octoprint server an start the print fro the LCD there, so I don't need to have my computer every single time I want to start a print.

marcopiraccini, 2019-06-20:
I don't think this is actually possible. But if you want this, why don't you simply copy the gcode files to the SD?

tbedore, 2019-07-03:
In addition, it is impossible to stop a print from the printer if you are printing via Octoprint.  You have to either hit the x button which disconnects the printer from Octoprint, or you have to go to your computer and cancel the print from there.  I should be able to just hit cancel print like I do with my SD card

Mjvdg, 2019-07-23:
If the prusa can add start, stop, pause  to the menu, when a macro activation command (M..) is send to the prusa and then only send actions to the octoprinter. when one of the commands is given. 
u can use the action plugin of octoprint to stop,start and pause it.  this would be a great feature

leptun, 2019-09-29:
@tbedore Here is a quick demonstration of the feature I am working on right now. It solves the Printing from SD from octoprint in a very unconventional way.
It's still in development though.
I haven't changed anything regarding Stopping and Pausing prints from the LCD, but this is a start. Right now I am also waiting for @foosel to answer to the issue I posted regarding the broken formatting.
I believe this feature will make using flashair something extremely easy and might make usb printing for those people who have it obsolete since printing from sd has a performance advantage.
https://photos.app.goo.gl/oca4R7jY6qHqEVKE9

closed 2023-09-28 by github-actions[bot]