prusa3d/Prusa-Firmware #1929: Turning MK3 Off From Outlet

tbedore, 2019-06-17:
I use a WeMo switch connected directly to the power on my MK3 so if there is a problem I can kill the power to the printer so it doesn't make the problem worse.  

I recently updated to the 3.7.2 RC1 and I have noticed that whenever I do this, there is a loud beep sound.  Probably meaning the power panic is being triggered.  I don't think it has ever done this before.  Not sure if it is a bug.  But I attached a video of the sound.  It is whenever you turn off the printer.  Doesn't matter if it is running a print or not.  

[IMG_2576.mp4.zip](https://github.com/prusa3d/Prusa-Firmware/files/3299195/IMG_2576.mp4.zip)

F0x06, 2019-06-18:
Had the same issue if i turn off the power with the PSU switch.
3.7.2 RC1

F0x06, 2019-06-28:
Still not fixed in 3.7.2 Release...

Rebel154, 2019-06-29:
I have the issue as well. Flashed 3.7.2 Final yesterday and now get a beep on printer switch off and my stats for power failure increment by one each time.

frakman1, 2019-07-02:
I noticed this too on my MK3S (3.7.2 release). I turn mine on/off with a Voice Activated Smart Plug and beeps everytime I turn it off even when it's sitting idle.

This is definitely a bug. It shouldn't beep loudly or complain in any way if I am not printing anything.

Canceltodebug, 2019-07-05:
Same issue. Raised with support and they have confirmed the issue.

F0x06, 2019-07-05:
> Same issue. Raised with support and they have confirmed the issue.

Thanks for the info, Prusa must reply more frequently to github issues, Just to let know users that they are working on it.

barthap, 2019-07-13:
Same issue here on my new Mk3s. Power panic beeps even if printer is turned off using PSU switch. 
Good to know that it might be software issue.
I don't know if this happens since I updated FW to 3.7.2, because I updated it immediately after assembling my kit.
I've also checked power panic connectors & wire between PSU and Einsy Rambo board, but they're all right.

Canceltodebug, 2019-07-13:
You can install 3.7.1 to get rid of the noise. I imagine 3.7.3 will be released this month.

leptun, 2019-07-13:
It should be fixed in 3.8

ochm, 2019-07-22:
On MK3 with oild power supply (silver) no problem with powerpanic in any firmware version.
On MK3S with new power supply (black) is powerpanic (in printer idle state) detected in FW3.7.1 and 3.7.2 too always the printer is turned off (or disconnected from power supply.. The Z is moved little bit up, printer beeps, then goes off.

tbedore, 2019-07-22:
@ochm Okay so that's a little interesting, I have the old (silver) MK3 PSU, and it's doing the same thing that you are getting with the black power supply.  It looks like the printer is triggering power panic whenever it is being shut off directly from mains power, which it should only do if a print is happening.

epicDISZ, 2019-07-25:
Can confirm, it happens whenever the printer doesn't get power it fires power panic, even though it should not be firing power panic whenever the printer isn't printing. Seems like an issue with this commit, [https://github.com/prusa3d/Prusa-Firmware/commit/ff3c9d1a7cf648994648aa3843fbb12920755c18](https://github.com/prusa3d/Prusa-Firmware/commit/ff3c9d1a7cf648994648aa3843fbb12920755c18)
for 3.7.2.
Looks like something in the changes was incorrect and now the if statement is false even though it should be true. I tried using Pronterface, to turn off motors/power supply with gcodes, but still power panics.

tbedore, 2019-07-26:
Based on the commit it looks like they fixed the issue.  I’m going to still keep this thread open unless prusa themselves closes it, until we get the next release candidate and the issue is no longer present.

dav991, 2019-08-09:
On my mk3s with 3.8.0 RC1  the issue is not reproducible.

frakman1, 2019-08-17:
@dav991 How can I get 3.8.0 RC1 ?

leptun, 2019-08-17:
https://github.com/prusa3d/Prusa-Firmware/releases

Please note there are some small issues with the current release.

frakman1, 2019-08-18:
@leptun Thank you!

FloatingCameras, 2019-09-01:
3.8.0 RC2 - I have confirm that it works on my MK3S/MMU2S.

leptun, 2019-09-01:
@tbedore Two people have confirmed it’s fixed. I think the issue can be closed now.

closed 2019-09-01 by tbedore