prusa3d/Prusa-Firmware #1931: New language submission

uniacke1, 2019-06-18:
I am willing to make a translation to offer support for an additional language (NL). 

However, the documentation for this is not very clear. It looks like I either need to submit just a new lang_??.txt or a .po file for the language. 

Is this correct? If not, what else is required?

Thanks!

3d-gussner, 2019-06-18:
I tried few times to get other languages into Prusa firmware they never pulled my pull requests.

You can checkout my https://github.com/3d-gussner/Prusa-Firmware/tree/MK3-lang_nl (not up2date).  
Compare my branch to Prusa MK3 branch and you can see the differences.

Here the PR i made to Zaribo firmware https://github.com/Zaribo/Zaribo-Firmware/pull/14

More up2date version you can find here https://github.com/Zaribo/Zaribo-Firmware. 

The .po files can be 
- updated using lang-export.sh 
- imported using lang-import.sh

Most important is that the all lang_en_??.txt have the same amount of translations. 
Example:
If lang_en.txt has 364 entries
all other need to have 364 entries.

If there is one missing, you can get strange behavior on the LCD and just a factory reset helps.

uniacke1, 2019-06-18:
I was surprised there wasn't one yet... this explains it. 
 
Never mind then, I guess - if you've already done the legwork and had it refused, I don't see why my efforts would be any different.

closed 2019-06-18 by uniacke1