prusa3d/Prusa-Firmware #1936: Can't change language (previous firmware was single language)

boraca, 2019-06-19:
My previous firmware was 3.7.1 single language build. Now I'm on official 3.7.2-rc1 from here (multi-language). When I go to menu to change language and select it I'm asked if I want to copy over the new language and no matter what I choose I'm brought back to language selection menu and language stays on English. I tried Polish at first, but all languages seem affected. I use English anyway, but the email ask to verify my language so that's where the issue is coming from.

I can send a video or terminal output if it's needed.

leptun, 2019-06-19:
You should try clearing all the settings from eeprom.

boraca, 2019-06-22:
I did that, and it didn't help, but now some messages are in Polish, with language set to English.
For example, with MMU disconnected, when filament sensor is triggered it says: "Load filament 4" in Polish, then some gibberish mixed with English. Rest of the interface is in English and works fine.

leptun, 2019-06-22:
Did you try reflashing the firmware? There was a similar issue and it was resolved by this. Maybe the fix also works on this version. 
Issue: #1831
https://github.com/prusa3d/Prusa-Firmware/issues/1831

boraca, 2019-06-27:
OK, here's what I did:
1. Flashed 3.7.2-RC1 again - problem persists
2. Flashed 3.7.0 - language selection works fine
3. Flashed 3.7.2-RC1  - it asks to select language on first boot, but the error is still there - can't select anything other than English.

If that's any help, here's startup messages from those firmwares:

`Recv: boot_src_addr  =0xc2400000

Recv: boot_dst_addr  =0x11108d06

Recv: boot_copy_size =0xeda2
  
Recv: boot_reserved  =0x70

Recv: boot_app_flags =0x80

Recv: boot_app_magic =0x55aa55aa

Recv: \x00echo: 3.7.0-2201`

and 3.7.2-RC1:
`Recv: boot_src_addr  =0x00001044

Recv: boot_dst_addr  =0x00002500

Recv: boot_copy_size =0x0000

Recv: boot_reserved  =0x70

Recv: boot_app_flags =0x80

Recv: boot_app_magic =0x55aa55aa

Recv: \x00echo: 3.7.2-RC1-2359`

3.7.0 has 2073 bytes free memory on startup, 3.7.2 2013 bytes.

leptun, 2019-07-15:
Yes, multilanguage support is broken for me also on two boards. Will try official fw soon

@DRracer Please look into this. I need to test some lcd translations but cannot because of this

DRracer, 2019-07-15:
Just checked an MK3 with the stock FW and Czech and Polish language and found no problem. What is your machine?

leptun, 2019-07-15:
I was trying on a compiled MK2 from MK3 branch since it was the latest, but without many features enabled. Just tried with MK2.5 definition and official fw from the site works, but self compiled firmware doesn't.

Here is compiled output:
[FW372-RC1-Build2359-1_75mm_MK25-RAMBo13a-E3Dv6full.zip](https://github.com/prusa3d/Prusa-Firmware/files/3393167/FW372-RC1-Build2359-1_75mm_MK25-RAMBo13a-E3Dv6full.zip)

It behaves exactly like @boraca said in initial post.

boraca, 2019-07-15:
@DRracer My printer is MK3S upgraded from MK3. Final 3.7.2 has the same problem.

DRracer, 2019-07-16:
@leptun could you try to build from an older release (e.g. 3.7.2)? Just want make sure we fix all the build issue that come with the new board definition for the upcoming 3.8.

leptun, 2019-07-16:
@DRracer Compiled 3.7.2 release and it works on multi-language. Wondering if some new change broke it or weather my local repo has some ignored files causing problems.

leptun, 2019-07-18:
@DRracer @boraca Fixed it for me. In my case it was the local repo that was broken in WSFL because of LFCR issues with the source files. After running dos2unix on all files of type "*h" "*c" "*cpp" "*txt" "*sh". The problem was fixed.

leptun, 2019-07-21:
@boraca Did you get it sorted out?

kgolger, 2019-07-22:
the same problem with my original MK3S/MMU2S: change language from english to german works in original FW 3.7.0. and older.
3.7.1 and 3.7.2 are not working. I tried all: full factory reset several times, flashed with newest Prusa Slic3r 2.0.0. Tried to recompile FW with arduino (build.sh), multilanguage=on (1) after ran a dos2unix over all -c .ccp .h .txt .py and .sh.
Flashing, flashing again but language change does not work. LCD is blinking 3 times and fall back to the language menu.
As is said: 3.7.0 an older are working without any problems.
Checked all hardware components -> no broken wires, all ok!

leptun, 2019-07-22:
@kgolger Did you also try a precompiled version from prusa, or you only tried versions compiled by you?

boraca, 2019-07-22:
@leptun precompiled versions of 3.7.2 and 3.7.1 have the exact same issue.

leptun, 2019-07-22:
I'm not sure what might cause that. Can you please send the log from the firmware updater utility and another one that contains the messages received from the printer over usb when changing the language.

boraca, 2019-07-22:
Sure, when I get home. Meanwhile I noticed that:

    #define LANG_SIZE_RESERVED     0x2800 // reserved space for secondary language (10240 bytes)

was reduced from 12032 bytes between those releases in config.h. Perhaps one of the languages doesn't fit in the new reserved size?

leptun, 2019-07-22:
I'm not sure if that's what's happening since I was able to compile 3.7.2 and use precompiled 3.7.2 without issues.

kgolger, 2019-07-22:
i tried the original unchanged download hex-files from prusa drive side for MK3S/MMU2S FW 3.7.2/1.05 and i also tryed to compile the original source-files with build.sh on linux ubuntu 18.04 LT.
Here is the log-file from build.sh:

HEX-Dateien werden geschrieben -> /home/admin/Schreibtisch/MK3-Firmware/Prusa-Firmware-3.7.2/lang/
und (NAS-DISKSTATION) -> /run/user/1001/gvfs/smb-share:server=192.168.0.17,share=raspberry,user=admin/Prusa_Mk3_Firmware_Flashfiles/

Log-Datei -> /home/admin/Schreibtisch/build.log
kgolger - 08.07.2019

/home/admin/Schreibtisch/MK3-Firmware/Prusa-Firmware-3.7.2/build.sh gestartet:

Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 250326 bytes (98%) of program storage space. Maximum is 253952 bytes.
Global variables use 6142 bytes of dynamic memory.
lang-build.sh started
lang=en
count=311
offs=638
size=8838
lang=cz
[E]: Text "\x1b[2JPOZOR:\x1b[1;0HCrash detekce\x1b[2;0Hdeaktivovana ve\x1b[3;0HStealth modu" is longer then definiton on line 17
[E]: Text "Dostavovani Z:" is longer then definiton on line 25
[E]: Text "a stisknete tlacitko" is longer then definiton on line 41
[E]: Text "SpoolJoin   [zap]" is longer then definiton on line 49
[E]: Text "SpoolJoin   [vyp]" is longer then definiton on line 57
[E]: Text "AutoZavedeni fil." is longer then definiton on line 65
[E]: Text "Automaticke zavadeni filamentu je dostupne pouze pri zapnutem filament senzoru..." is longer then definiton on line 69
[E]: Text "Automaticke zavadeni filamentu aktivni, stisknete tlacitko a vlozte filament..." is longer then definiton on line 73
[E]: Text "Korekce E:" is longer then definiton on line 217
[E]: Text "Vysunout filament" is longer then definiton on line 221
[E]: Text "Kon. spinac nesepnut" is longer then definiton on line 233
[E]: Text "F. autozav. [zap]" is longer then definiton on line 273
[E]: Text "F. autozav. [N/A]" is longer then definiton on line 277
[E]: Text "F. autozav. [vyp]" is longer then definiton on line 281
[E]: Text "Rychlost vent." is longer then definiton on line 289
[E]: Text "Kontr. vent.[zap]" is longer then definiton on line 297
[E]: Text "Kontr. vent.[vyp]" is longer then definiton on line 301
[E]: Text "Filament nezaveden" is longer then definiton on line 325
[E]: Text "Spotrebovano filamentu" is longer then definiton on line 333
[E]: Text "Kal. prvni vrstvy" is longer then definiton on line 349
[E]: Text "Predni tiskovy vent?" is longer then definiton on line 369
[E]: Text "Levy vent na trysce?" is longer then definiton on line 505
[E]: Text "Predehrejte trysku!" is longer then definiton on line 785
[E]: Text "RPi port    [zap]" is longer then definiton on line 889
[E]: Text "RPi port    [vyp]" is longer then definiton on line 893
[W]: No display definition on line 9
[W]: No display definition on line 21
[W]: No display definition on line 37
[W]: No display definition on line 53
[W]: No display definition on line 61
[W]: No display definition on line 77
[W]: No display definition on line 81
[W]: No display definition on line 85
[W]: No display definition on line 89
[W]: No display definition on line 93
[W]: No display definition on line 97
[W]: No display definition on line 105
[W]: No display definition on line 117
[W]: No display definition on line 121
[W]: No display definition on line 125
[W]: No display definition on line 145
[W]: No display definition on line 149
[W]: No display definition on line 153
[W]: No display definition on line 157
[W]: No display definition on line 161
[W]: No display definition on line 165
[... 1122 lines cut ...]
config.sh finished with success
postbuild.sh started
 checking files:
  folder  OK
  elf     OK
  objects OK
 running progmem.sh...OK
 running textaddr.sh...OK
 checking textaddr.txt...NG! - some texts not found in lang_en.txt!
  missing text ignored!
 extracting binary...OK
 updating binary:
  primary language ids...OK
  primary language signature...OK
 converting to hex...OK
  secondary language data...Updating languages:
 Czech  : OK
 German : OK
 Italian: OK
 Spanish: OK
 French : OK
 Polish : OK
 converting to hex...OK

postbuild.sh finished with success

leptun, 2019-07-22:
@kgolger I am more interested in the uploading and language changing than in compiling. One thing for sure is that the compiled output looks ok.

kgolger, 2019-07-22:
**here is the avrdude_output_log file when i flashed 3.7.2:**

avrdude: Version 6.3-20160220-prusa3d, compiled on Feb 11 2019 at 18:25:11
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:/PROGRA~1/Prusa3D/Slic3rPE/resources\avrdude\avrdude.conf"

         Using Port                    : COM5
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: safemode: hfuse reads as D0
avrdude: safemode: efuse reads as FD
avrdude: reading input file "C:\Users\7800mbz\Desktop\firmware.hex"
avrdude: writing flash (250326 bytes):

Writing | ################################################## | 100% 43.99s

avrdude: 250326 bytes of flash written
avrdude: verifying flash memory against C:\Users\7800mbz\Desktop\firmware.hex:
avrdude: load data flash data from input file C:\Users\7800mbz\Desktop\firmware.hex:
avrdude: input file C:\Users\7800mbz\Desktop\firmware.hex contains 250326 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 32.06s

avrdude: verifying ...
avrdude: 250326 bytes of flash verified

avrdude: safemode: hfuse reads as D0
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude done.  Thank you.

avrdude: Version 6.3-20160220-prusa3d, compiled on Feb 11 2019 at 18:25:11
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:/PROGRA~1/Prusa3D/Slic3rPE/resources\avrdude\avrdude.conf"

         Using Port                    : COM5
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: prusa_init_external_flash(): MK3 printer did not boot up on time or serial communication failed
avrdude: arduino_open(): Failed to initialize MK3 external flash programming mode

avrdude done.  Thank you.

_______________________________________________

**avrdude.conf is untouched!**

leptun, 2019-07-22:
I think you just got the answer why multiple labguages don’t work:

avrdude: prusa_init_external_flash(): MK3 printer did not boot up on time or serial communication failed
avrdude: arduino_open(): Failed to initialize MK3 external flash programming mode

Please take a photo of the board. I want to know if you have the flash chip on the board. You might also have it on a daughterboard that shifts the lcd connectors.

If you look at the schematic it’s U13. 
https://github.com/ultimachine/Einsy-Rambo/blob/1.1a/board/Project%20Outputs/Designators_Einsy%20Rambo_1.1a.PDF

https://github.com/ultimachine/Einsy-Rambo/blob/1.1a/board/Project%20Outputs/Schematic%20Prints_Einsy%20Rambo_1.1a.PDF

leptun, 2019-07-22:
@kgolger 
This is a board that doesn't have a SPI flash chip for languages:
[image]

In that case Prusa ships the board with a daughterboard that contains the required chips:
[image]

On newer boards the SPI flash is directly on the board, requiring no external chip:
[image]

Which of these is your case?

boraca, 2019-07-22:
Here's the flashing log: https://gist.github.com/boraca/800c9b82dfe6312f4b194db303e1b13a
My board doesn't have the daughterboard.

And here's output when switching to Czech:
```
Recv: boot_src_addr  =0x01fe1efb
Recv: boot_dst_addr  =0xc2400000
Recv: boot_copy_size =0x2189
Recv: boot_reserved  =0x20
Recv: boot_app_flags =0x80
Recv: boot_app_magic =0x55aa55aa
Recv: \x00echo: 3.7.2-2363
```
and German:
```
Recv: boot_src_addr  =0x00000e6a
Recv: boot_dst_addr  =0x00002400
Recv: boot_copy_size =0x0000
Recv: boot_reserved  =0x30
Recv: boot_app_flags =0x80
Recv: boot_app_magic =0x55aa55aa
Recv: \x00echo: 3.7.2-2363
```

3d-gussner, 2019-07-22:
Maybe stupid question, but where did you get the EINSY from? As Prusa has it own bootloader to squeeze some more bytes for the firmware it will not fit on an Ultimachine EINSY.

kgolger, 2019-07-22:
this is mine:

[image]

as i said: till 3.7.0 i had never problems with change language.
i think, if i would have no flash memory, the older versions would also not work, but they did.

leptun, 2019-07-22:
@boraca In your case it appears that German is corrupted, but czech is not (look at the size). Is that correct? Is your upload successful?

@kgolger Please do exactly like boraca did and also confirm the spi flash type.

3d-gussner, 2019-07-22:
@kgolger Another stupid question, what happens if you try to upload an official hex file?

leptun, 2019-07-22:
@kgolger that is true.
@3d-gussner actually fair point.

leptun, 2019-07-22:
@3d-gussner he said he tried from the website.

3d-gussner, 2019-07-22:
Full factory reset?

3d-gussner, 2019-07-22:
BTW. Github has some issues giving me and others 500 error but we can still work in this issue :-)

leptun, 2019-07-22:
@3d-gussner Saw that too...🤷‍♂️

boraca, 2019-07-22:
Czech version doesn't work even if the flashing succeeded. Here's the video: https://www.youtube.com/watch?v=e-xVn8yHwrA

Only way to exit the menu is to select English.

3d-gussner, 2019-07-22:
@kgolger Where did you get this board? It is green and if I am not wrong Prusa EINSy are black (maybe the 1st where green not sure about that)
@boraca Can you post a picture of your board? Is it a Original Prusa EINSy? And did you try factory reset?
https://help.prusa3d.com/article/SYvbQ66IXF-factory-reset "All data"

boraca, 2019-07-22:
@3d-gussner Sure: https://imgur.com/a/TmH5eO5 It's an Original Prusa MK3, upgraded to MK3S, MMU2S is disconnected.

3d-gussner, 2019-07-22:
@boraca Can you compile your own Firmware? If so please download https://github.com/prusa3d/Prusa-Firmware/archive/v3.7.2.zip and uncomment following lines in Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h

```
//#define DEBUG_SEC_LANG   //secondary language debug output at startup
//#define DEBUG_W25X20CL   //debug external spi flash
```
to
```
#define DEBUG_SEC_LANG   //secondary language debug output at startup
#define DEBUG_W25X20CL   //debug external spi flash

```

use `build.sh` or `PF-build.sh` to compile the firmware

open Pronterface and see what you get via serial while trying to change the language

3d-gussner, 2019-07-22:
@boraca  Another stuipid question? How do you update the firmware? Do you use Prusa Slicer 2.x or do you use something else?

3d-gussner, 2019-07-22:
As far i know EINSy English_only firmware hex files can be uploaded via Prusa Slicer, Arduino IDE, OctoPrint Firmware updater plug in and many other.
Multiple languages miniRambo MK2.5 hex files can be updated the same, as the 2nd language is stored in the EEPROM.

Multiple languages EINSY hex files to my knowledge just works with Prusa Slicer Firmware uploader as additional languages are stored in external flash.

leptun, 2019-07-22:
You are partially right. The second language on mk2.5(s) is stored in the internal flash alongside the firmware. The eeprom is only used for storing settings

kgolger, 2019-07-22:
i bought my original Prusa MK3 in June 2018 and made no changes on board or otherwhere. i always have download the firmware from the original prusa driver download side. since june 2018 i have regulary updated the Prusa FW, till 3.7.0 and i had never problems with language select. mayby there is something with the board? but why does 3.7.0 work??
i have always updated the FW MK3/S und MMU2/S with the neweset Slic3r PE.

boraca, 2019-07-22:
I get this error when compiling with those 2 lines uncommented (normally compiles fine):
```
In file included from sketch/language.c:2:0:
language.h:135: error: unknown type name 'FILE'
 extern uint16_t lang_print_sec_lang(FILE* out);
                                     ^
sketch/language.c: In function 'lang_get_sec_lang_str_by_id':
sketch/language.c:234:2: warning: return makes pointer from integer without a cast
  return ui + pgm_read_word(((uint16_t*)(ui + 16 + id * 2))); //read relative offset and return calculated pointer
  ^
exit status 1
```

I'm using Octoprint with FirmwareUpgrade plugin. I will try with PrusaSlicer.

leptun, 2019-07-22:
@boraca I’m not sure why you get the error. Did you set everything according to the instructions in readme.md?

Also, it might not matter after all since you said you uploaded with octoprint. That’s the problem. The firmware updater in prusaSlicer is a custom version of avrdude that has the support to also push to the external flash of the rambo. If you use regular avrdude you can only push the main program, which looks for the languages in the flash and finds they are wrong in some way since it’s from an older version.
Use PrusaSlicer to update and you should be good.

boraca, 2019-07-22:
Yeah I followed readme.md and compiled firmware before. The error happens if I uncomment those lines:
```
#define DEBUG_SEC_LANG   //secondary language debug output at startup
#define DEBUG_W25X20CL   //debug external spi flash
```
per @3d-gussner  instructions.

closed 2019-07-22 by boraca

kgolger, 2019-07-23:
SOLVED!!!
I installed the newest Slic3r 2.0.0 on a Ubuntu 18.04 LT machine, started flashing with Slic3r and the result:
Bingo! No errors, switching to another language is no problem.
Here are the logs of 1.st working Ubuntu.log and 2.nd not working W7.log:

**Ubuntu 18.04 LT working:**

Prusa Slic3r (Ubuntu 18.04 LT) result:

avrdude-slic3r -v -p atmega2560 -c wiring -P /dev/ttyACM0 -b 115200 -D -U flash:w:0:/home/admin/Schreibtisch/MK3-Firmware/Prusa-Firmware-3.7.2/lang/firmware.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on May 20 2019 at 18:53:06
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : /dev/ttyACM0
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "/home/admin/Schreibtisch/MK3-Firmware/Prusa-Firmware-3.7.2/lang/firmware.hex"
avrdude-slic3r: writing flash (250326 bytes):

Writing | ################################################## | 100% 44.01s

avrdude-slic3r: 250326 bytes of flash written
avrdude-slic3r: verifying flash memory against /home/admin/Schreibtisch/MK3-Firmware/Prusa-Firmware-3.7.2/lang/firmware.hex:
avrdude-slic3r: load data flash data from input file /home/admin/Schreibtisch/MK3-Firmware/Prusa-Firmware-3.7.2/lang/firmware.hex:
[... 153 lines cut ...]
avrdude: 250326 bytes of flash verified

avrdude: safemode: hfuse reads as D0
avrdude: safemode: efuse reads as FD
avrdude: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude done.  Thank you.

avrdude: Version 6.3-20160220-prusa3d, compiled on Feb 11 2019 at 18:25:11
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:/PROGRA~1/Prusa3D/Slic3rPE/resources\avrdude\avrdude.conf"

         Using Port                    : COM5
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: prusa_init_external_flash(): MK3 printer did not boot up on time or serial communication failed
avrdude: arduino_open(): Failed to initialize MK3 external flash programming mode

avrdude done.  Thank you.

________________________________________________
That's it!
Thanks to all for their time und tipps!

3d-gussner, 2019-07-23:
@kgolger Glad it works.