prusa3d/Prusa-Firmware #1938: Unsuccessful filament loading after IR sensor positive detection

castayuelas, 2019-06-19:
After 3.7.1 version mmu2 filament loading into the nozzle has very low percentage of success. After load the filament into the nozzle and get a positive IR sensor detection the filament is retracted, mmu2 try to load again the filament with the same result geetting a message that mmu2 required attention. Usually , after intervention in the mmu2 but not doing any change , the mmu2  try to load the filament into the nozzle with success.

The only way I found to solve this in uploading version 3.6.0 again and I am able to print again without issues.

Demolux, 2019-06-19:
You can change the retraction distance in mmu.cpp, see here:

https://github.com/prusa3d/Prusa-Firmware/issues/1935

However, the given values should be ok with an unmodded printe/hotend and if the filament was removed via unload beforehand. If nothing is stuck in the e3ds filament path, this cannot be the issue.

Instead, look at your idler and the IR-sensor calibration. The idler has to rotate quite freely and without any wobble (I inserted a brass tube for the screw-joint to remove wobble). The filament sensor maybe setup so tight that it detects a "no filament" just by bondtech-gear wobble.

Kylra, 2019-07-08:
Decreasing the retract distance in mmu.cpp solved this issue. 52 was too high.
It unloaded the filament.

static bool can_load()
current_position[E_AXIS] += 60.0;
current_position[E_AXIS] -= 46.0; // old 52.0

closed 2023-09-28 by github-actions[bot]