prusa3d/Prusa-Firmware #1941: load to nozzle failure can result in blank LCD

PxT, 2019-06-22:
Machine:  MK3S + MMU2S
Firmware: 3.7.1-2266
MMU firmware: 1.0.5-297

Steps to reproduce:
* Preheat
* Load to nozzle -> Select filament
* Filament load fails for some reason (IR sensor doesn't see it, or filament doesn't engage gears properly, any reason at all for the purposes of this bug).  MMU2 goes into fail state with amber light flashing as expected
* Push 3rd button on MMU to retry the load
* LCD goes completely blank
Pushing the selector button takes the printer back to the main menu, so this is not a critical problem, but I think it is a bug.

fopeczek, 2020-03-03:
I have similar bug.
But in my printer the LCD is all right.

fopeczek, 2020-03-03:
I think to repair this bug you must somehow move the chimney with IR sensor...
[image]
or edit the chimney to allow him to move freely on x axis.

gudnimg, 2023-07-27:
Let's close this issue after 3.13.0 is released.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg