prusa3d/Prusa-Firmware #1949: Load Filament 256 1st Layer Calibration in Setup Wizard

jsilva-iii, 2019-06-28:
MK2.5S+MMU2S - FW 3.7.2

After bed calibration in the setup wizard, one is prompted to load filament 1 and press ok - the mmu2S goes through it's initialization steps, then selector goes to the 5th filament and a message appears on the screen "Load filament 256" and if you inset filament in the 5th guide it won't load
[image]

jboymel, 2019-06-28:
This would sometimes happen with MK3S+MMU2S - FW 3.7.1 as well.

karrosen, 2019-07-23:
Just got this issue myself, just flashed to FW 3.7.2

(All the while I'm trying to solve another issue entirely.)

crioux, 2019-08-19:
Also having this issue wit FW3.7.2

AndyNewlands, 2019-08-31:
I have the exact same issue with MK3S FW 3.7.2-2363 & MMU2S FW 1.0.6-372

Also, if you persevere will the 5th guide, after a reboot, and get to it accept there's filament present, it will eventually start trying to pull filament for the Z-calibration from hole number ONE! (in which there's no filament.  Infuriating ;-)

gudnimg, 2023-07-24:
Let's close this issue when 3.13 is released 👍

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg