prusa3d/Prusa-Firmware #1951: system specific envirnment variable use -- replace __linux__ checks with WXGTK or __unix__ where possible

GaryAitken, 2019-06-28:
Was requested to continue this discussion here; original discussion is at
  https://forum.prusaprinters.org/forum/english-forum-general-discussion-announcements-and-releases/help-porting-slicer-wxwidgets-environment-variables/#post-149639
Relevant portions of that are repeated below.
See also references in this link
  https://lists.freebsd.org/pipermail/freebsd-ports/2019-June/116572.html
I'm trying to port the slicer to FreeBSD, encountered issues with environment variables.
The code is full of things like:
  #if defined \_WIN32
    ...
  #elif defined(\_\_linux\_\_)
    ...
  #elif defined \_\_APPLE\_\_
    ...
occasionally there is a
  #if defined(\_\_NetBSD\_\_) || defined(\_\_OpenBSD\_\_) || defined(\_\_bsdi\_\_) || defined(\_\_DragonFly\_\_)
or a
  #ifdef \_\_unix\_\_
or a
  #ifdef \_\_WXGTK\_\_    (for wxwidgets toolkit mapped to gtk)

It looks like many of the uses of \_\_linux\_\_ are actually gtk issues which it seems should be common to linux and bsd under gtk on X.  In these cases, is there some reason \_\_linux\_\_ was used and not \_\_WXGTK\_\_? If one keeps the \_\_linux\_\_ tests in these cases, it's necessary to add "|| defined(\_\_FreeBSD\_\_) || defined(\_\_NetBSD\_\_)..." etc.  

> At the UI level, we likely could use \_\_WXGTK\_\_. we don't link the wxWidgets library to the non-ui modules, so some other macro should be used there.

ok, that's helpful.  Using \_\_WXGTK\_\_ would greatly simplify porting to other unix-like systems.

Is there a "BSD" sym that encompasses the various bsd variants?

> I don't know. We will be glad if you contribute with your investigation.
There is not, at least not a useful one.

If \_\_linux\_\_ is defined, is \_\_unix\_\_ also defined?  If any of the \_\_*BSD\_\_ syms are defined, is \_\_unix\_\_ also defined?

> Again, we don't know. Please investigate.

The double underscored syms are reserved and generated by the compiler.  In all cases for unix-type systems except mac, \_\_unix\_\_ is also defined.  So for linux, freebsd, openbsd, etc., \_\_unix\_\_ will also be defined.  Despite the fact that later versions of mac-os are bsd based, \_\_unix\_\_ is not defined for them.

So it seems instead of using \_\_linux\_\_ in much of the code, \_\_unix\_\_ would be preferable if the condition is generally unix related, not specific to linux, and not a ui/wx-widgets/gtk issue.

What's the relationship between the \_WIN32, \_\_APPLE\_\_, \_\_unix\_\_ syms in code and the WIN32, APPLE, UNIX, XCODE syms in the CMakeLists.txt file?  Are any/all of these case sensitive? How is CMAKE_SYSTEM_NAME, which may be "Linux" (on linux), related to the above syms?

> Again, I would have to look up. I think the underscored symbols are intrinsically generated by the compiler. The others are either generated by the CMake, but they as well may be defined by some include.

The syms are case-sensitive if the underlying system is case-sensitive.  CMAKE_SYSTEM_NAME is extracted from uname -s (or similar) output, and may well be different from the \_\_xxx\_\_ system name (e.g. "Linux" instead of \_\_"linux"\_\_).  The all caps syms used in CMakeLists.txt are generated and not directly related to or derivable from the \_\_xxx\_\_ system syms.

I would like to propose that wherever possible in ui code, \_\_linux\_\_ be replaced by \_\_WXGTK\_\_; in non-ui code \_\_linux\_\_ be replaced with \_\_unix\_\_; and only where something is specific to linux should \_\_linux\_\_ be used.  I believe changing all \_\_linux\_\_ to \_\_unix\_\_ should be possible without consequences on linux, mac, and windows builds.  Someone trying to port to some other system such as freebsd would then need to be concerned only with things where linux differs from the other system, which are hopefully few.

GaryAitken, 2023-09-20:
It's stale only because no-one at Prusa has dealt with it, unfortunately.  It seems inadvisable to close it because it is an open, fairly easily resolved, issue -- make the changes suggested in the last paragraph, and see if it works on the platforms you support.  Something relatively easy to do, and easy to back out, in an alpha release.  I suspect these are changes that should be pushed upstream also.

gudnimg, 2023-09-20:
The mention of Wxwidgets makes me suspect this should be raised in the PrusaSlicer repository. https://github.com/prusa3d/PrusaSlicer I’m not familiar with that codebase to judge if this is still relevant.

However this repository is only for the 8-bit firmware.

3d-gussner, 2023-09-21:
@GaryAitken As @gudnimg pointed out this issue should be reported in an other Repo.
Closing it.

closed 2023-09-21 by 3d-gussner