prusa3d/Prusa-Firmware #1954: Hotend and bed stay on after print finished

rackley096794, 2019-06-29:
MK3S / MMU2S.  After a MMU filament load failure and correction, the print proceeds as expected.  However, when the print completes the hot end and bed continue to stay set to the hot temperature.   This is with (current as of today) firmware versions 3.7.2 and 1.0.5.

vertigo235, 2019-06-30:
Can you share your GCODE, I have not experienced this

PatitoPatozo, 2019-06-30:
I also saw this once in 3.7.1 when the print ended with an unload failure (fat tip was stuck in the extruder).

vertigo235, 2019-06-30:
ahh maybe, so perhaps 

; Unload filament
M702 C

Does not timeout like a normal change command would, yes that would be a bug I think

wiktor-starzak, 2019-07-02:
I do have that issue too.
MMU: 1.0.5-297
Prusa: 3.7.1-2266
Slicer: 2.0.0+win64 (2019-07-02 09:19:58)

wiktor-starzak, 2019-07-03:
@rackley096794 I have contacted support and the did confirm the bug. Fix is on the way

ssvenn, 2019-08-12:
I just had this happen to me, didn't realize until the day after that the hotend and bed was still powered up. After I cleared the MMU2S error the LCD on the MK3S just keeps flickering. I reset the printer since it was finished with the print anyway.

closed 2023-09-28 by github-actions[bot]