prusa3d/Prusa-Firmware #1955: [3.7.2] Power Panic Beep When Turning Off From Wall Plug When Not Printing

sarf2k4, 2019-06-30:
I noticed something when I turned off the printer when on idle, not printing anything, I usually turns off my printer from wall plug directly because of easy access compared to the psu switch which is bit farther. The printer made power panic long beep when I turned off the wall plug where previous versions, no beep at all, is this a bug or a feature or does the firmware really make some data change to eeprom on this versions as well as previous ones about having power cut off from the mains plug?

F0x06, 2019-07-01:
Duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/1929

tbedore, 2019-07-26:
This issue is a duplicate of #1929, and was fixed in commit ff3c9d1

leptun, 2020-01-26:
@sarf2k4 Please close this issue.

closed 2020-01-27 by sarf2k4