prusa3d/Prusa-Firmware #1959: M80 not getting executed properly in certain circumstances.

tsprhdm, 2019-07-02:
While testing a new startup sequence, i stumbled upn an issue w/ 3.7.1 on MK3.

I have set Bed leveling mesh to 7x7, magnet compensation.

Standard config pre print code works fine, my code below does not. 

**Expected:**
Heat up to 50/170
home axes
_do 7x7 mesh bed level_
move fast to x0 y0 z50
heat up to PLA first layer ttemps

**observed:**
heat up to 50/170
homes axes
**dips to pinda once on front left corner**
moves up fast to x0y0 z50
heats up to PLA first layer temps.

See code here:

; generated by PrusaSlicer 2.0.0+win64 on 2019-07-02 at 11:41:33

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M73 P0 R50
M73 Q0 S51
M201 X1000 Y1000 Z1000 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E1.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M115 U3.7.1 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
M190 S50 ; wait for bed to come up to calibration temp
M109 S170 ; wait for nozzle to come up to calibration temp
G28 W ; home all without mesh bed level
M80 ; mesh bed level
G1 X0 Y0 Z50 F1000.0 ;
M190 S60 ; wait for bed temp
M109 S215 ; make sure nozzle is at correct temperature
G1 Y-3.0 F1000.0 ; go outside print area

leptun, 2019-07-02:
The command for mesh bed leveling is G80, not M80

You can also accomplish the same thing by just using G28 without the W at the end

tsprhdm, 2019-07-02:
D'oh.

Just checked, and: as soon as i do right, it does it right.

closed 2019-07-02 by tsprhdm