prusa3d/Prusa-Firmware #1961: Feature Request: LCD knob acceleration

haarp, 2019-07-03:
Turning the knob to scroll through large ranges, e.g. from 0°C to 250°C, takes a lot of time and effort. Each click of the knob corresponds to unit of change.

I think acceleration would be a nice addition to the knob. Turning it slowly should continue to step through 1 unit at a time, but turning it at higher speeds should progressively step up the unit-per-click ratio.

Example:
Turning the knob 40 clicks (1 revolution) in 5 seconds -> 40 units are changed
Turning the knob 40 clicks in 1 second -> 160 units are changed

These obviously need fine-tuning. A similar concept has been successfully used for a long time on computer mouse acceleration.

Cheers!

ManuGithubSteam, 2019-07-07:
Very good idea. I second this.

maximlevitsky, 2019-07-07:
I too would like that implemented.

GWdd, 2019-07-28:
Since most acceleration schemes fail for the stupidest reasons, why not make the increment of large scale items like temperature steps of 5 or 10 degrees.   In 99% of cases, you don't need single digit precision.  

Or, do a Press and Hold to increment digit selection. Default to 10 degrees per step, press and hold to rotate through the digits, cursor shows which digit is being modified.

romeoalfa1975, 2021-06-01:
Ich fände es schon hervorragend, wenn das Menü bei jedem Klick des Drehknopfes überhaupt reagieren würde. Manchmal reagiert es gar nicht, gelegentlich springt es sogar in die verkehrte Richtung...

3d-gussner, 2021-06-01:
> Ich fände es schon hervorragend, wenn das Menü bei jedem Klick des Drehknopfes überhaupt reagieren würde. Manchmal reagiert es gar nicht, gelegentlich springt es sogar in die verkehrte Richtung...

Please write in English as not everyone understands German.

romeoalfa1975, 2021-06-01:
I would like the menu to react at all with every click of the knob. Sometimes it doesn't react at all, sometimes it even jumps in the wrong direction...

closed 2023-09-28 by github-actions[bot]