prusa3d/Prusa-Firmware #1964: MK2.5/S Firmware 3.7.2 infinite unload/load loop when printing

tjhawkey, 2019-07-04:
My printer gets stuck in an infinite unload/load cycle when trying to print.
It preheats, mesh bed levels, but then unloads the filament before printing. 
When I unload and re-load, it goes through the mesh bed leveling and asks for an unload again.
The built in first layer calibration works fine.

leptun, 2019-07-04:
Please post the gcode which generates the problem.

leptun, 2019-07-04:
Also test that the filament sensor is working by going to “Main menu”-> “Support”->”Sensor info”. Check with the filament unloaded that the value changes corectly when inserting the tip of the filament in the extruder. If it shows a constant 0, then it is always senses as no filament present, hence the filament unload and reload.

tjhawkey, 2019-07-04:
Du-oh. The filament sensor isn't working. I'll track that down.
But, it would seem that if you have the sensor enabled and you say that the filament is loaded OK, that the firmware would point out the mis-match with the sensor state.
Thanks.

[1in calibration square_0.2mm_PLA_MK2.5S_14m.zip](https://github.com/prusa3d/Prusa-Firmware/files/3359573/1in.calibration.square_0.2mm_PLA_MK2.5S_14m.zip)

barthap, 2019-07-13:
I had the same issue, filmament sensor plug was loose.

closed 2023-05-26 by Prusa-Support