prusa3d/Prusa-Firmware #1965: Firmware doesn't compile with multiple_sheets merged.

leptun, 2019-07-05:
As the title says, it errors while compiling because of something related to eeprom.
Compiled on the MK3 branch with no modifications.

`C:\GIt\Prusa-Firmware\Prusa-Firmware-build\sketch\eeprom.h:221:1: error: non-constant condition for static assertion`
` static_assert(((sizeof(M500_conf) + 20) < EEPROM_LAST_ITEM), "M500_conf address space conflicts with previous items.");`

leptun, 2019-07-05:
Also at line 191 in eeprom.h shouldn't `EEPROM_NOZZLE_DIAMETER `be `EEPROM_NOZZLE_DIAMETER_uM `?

maximlevitsky, 2019-07-06:
Appears to compile here (linux, with build.sh script).

Note though that there is currently a cosmetic issue if you compile with language support - the multiple sheet selection menu has 2 letters missing from each menu - this is due the fact that the strings are passed to the translation function (_T()) but are not added to the translation. They will fix that for sure I think. Other than that it works superb.

leptun, 2019-07-06:
I am using PF-build.sh script. I am attempting to build on windows 10 with git bash. Did not enable multiple languages as I don't need them :)

vertigo235, 2019-07-06:
I built it fine, also included your LCD fix @leptun

vertigo235, 2019-07-06:
I also can confirm the issue of truncated menu items that @maximlevitsky reported, but so far the multi sheets seem to work I have calibrated 2 sheets already.

leptun, 2019-07-06:
Just tested on linux subsystem for windows with build.sh and PF-build.sh and it compiled on both of them. I suppose there is some difference between the linux and the windows variants of gcc.

leptun, 2019-07-06:
On windows the firmware compiles only with #1963 or #1934, but not both. It fails on the compile time checks added in #1934.

mkbel, 2019-07-09:
> Also at line 191 in eeprom.h shouldn't `EEPROM_NOZZLE_DIAMETER `be `EEPROM_NOZZLE_DIAMETER_uM `?

Fixed in https://github.com/prusa3d/Prusa-Firmware/pull/1982

mkbel, 2019-07-09:
> Appears to compile here (linux, with build.sh script).
>
> Note though that there is currently a cosmetic issue if you compile with language support - the multiple sheet selection menu has 2 letters [...]

Fixed in https://github.com/prusa3d/Prusa-Firmware/pull/1980

mkbel, 2019-07-09:
> As the title says, it errors while compiling because of something related to eeprom.
> Compiled on the MK3 branch with no modifications. [...]

Fixed in https://github.com/prusa3d/Prusa-Firmware/pull/1984

mkbel, 2019-07-09:
But anyway, Windows build is still expected to be broken by https://github.com/prusa3d/Prusa-Firmware/pull/1979 as this changed board definition. Our board definition is still work in progress as there is custom linker script missing and application size check fix missing https://github.com/DRracer/Arduino_Boards

leptun, 2019-07-09:
@mkbel Ouf... Back to LSFW then :'(

closed 2019-07-09 by leptun