prusa3d/Prusa-Firmware #1966: Ability to update printer statistics when printing via octoprint

salmander, 2019-07-05:
I liked that we have a statistics menu where we can see the total print time. But, the statistics don't get updated when printing via USB/Octoprint.

Can we please have printer update it's stats while printing via USB/Octoprint?

ManuGithubSteam, 2019-07-07:
I second this. :-)

haarp, 2019-07-07:
I think the problem is that the printer doesn't know whether movement through the serial interface is a print job or something else.

But I'm in favor of treating it as a print job regardless. I'd also like this.

salmander, 2019-07-07:
At the moment, Octoprint feels like a second class citizen with Prusa printers. It works for the most part but there are some things which Prusa has or does which are not compatible with Octoprint. This is one of those things. Kudos to Prusa for providing the ability to plug raspberry pi zero right into the motherboard and running Octoprint straight off it. But, that is also a dated solution. Raspberry pi zero is very slow for Octoprint and not even officially recommend by Octoprint itself. Now that raspberry pi 4 is out, it boots Octoprint almost as fast as the printer finished booting up. I think there is an opportunity here for Prusa. Sorry for my long rant. I love Prusa...love this community...love this printer. Thank you for everyone being involved in making it what it is.

ManuGithubSteam, 2019-07-07:
@salmander 
I know what you mean and stand behind you. I feel Prusa should officially sponsor octoprint to get better support.
However i feel they will relase the mk4 soon with new motherboard (Joe hinted that in one of his posts May i think.) And the new Mobo will have wifi so i imagine a solution direct from Prusaslicer to the printer. So they could ditch octoprint completely.

salmander, 2019-07-07:
My wife is not going to be happy with me if I say another printer 😄😄😄. But, yeah that sounds like a logical next step... Exciting times.

wavexx, 2020-02-26:
Related #676

drandreas, 2020-07-13:
I've implemented statistic support on top of M73. https://github.com/drandreas/Prusa-Firmware/commit/be3259b59b1562e0c959fd9596575d298f3ce1b1
It is by no means perfect but It might be of use to others. Just be warned of the risks involved in flashing a custom FW.

closed 2023-09-28 by github-actions[bot]